binsentsu/home-assistant-solaredge-modbus #2: Configuration through Integrationa Not possible

atze09, 2020-02-26:
Hello,
I successfully installed your component through hacs. After a restart i could Not find the configuration in Integrations.
I am on latest hassio build.
What did i wrong?
So i need to add Something to the configuration.yaml? If yes, what exactly?

thavini, 2020-02-26:
Hi,

You need to add something like below image.

Met vriendelijke groet/kind regards

Vincent

thavini, 2020-02-26:
[image]

binsentsu, 2020-02-27:
Hello,
After installation through hacs, check if the integration is installed under your custom_components folder. When this is the case, after a restart the integration should be visible within the 'Setup Integration' dialog (Configuration -> Integrations). There is no configuration in configuration.yaml needed.
[image]

atze09, 2020-02-27:
After updating hacs, deinstall solaredge component and completely fresh install the component i can see it in the intergrations.
Can i change/check the IP Port and frequency after installation.
At the moment the of every sensor is unknown, i already restartet the inverter.
Debug is empty.

binsentsu, 2020-02-27:
Changing IP and frequency after installation is not possible (this is at the moment the case for every home-assistant integration configured through config flow UI). But you can just remove the integration and re-add it again. If port and IP are correct and reachable the sensors should immediately update after interval has passed. If the sunspec registers are the same on all inverters it should work. I'm running SE3000HD without display on firmware 4.7.x. If you're unsure about IP/port or reachability within the network, I would advice to first test connection through a tool like https://sourceforge.net/projects/sockettest/ for example.

atze09, 2020-03-02:
Thank you for your hellp.
It was the same issue awulf was facing. Changed to solaredge Meter and Back to non se and now it works, very Strange.

Can you explain me what is the Advantage of your custom component in comparison with erikarenhill/solaredge-modbus-hass?

binsentsu, 2020-03-04:
Advantages:
- Configurable through UI. No yaml
- Separate sensor entities in home-assistant for each modbus key register. So no need for template sensors.
- User can disable entities in which he is not interested, through integration UI

atze09, 2020-03-04:
> Advantages:
> 
> * Configurable through UI. No yaml
> * Separate sensor entities in home-assistant for each modbus key register. So no need for template sensors.
> * User can disable entities in which he is not interested, through integration UI

Is the Data from the external Meter also available?
Amy disadvantages?

binsentsu, 2020-03-04:
External meter is currently not available. I only have the solaredge inverter at my place for testing. I haven't used the mentioned integration, only had a look at the code.

atze09, 2020-03-04:
Thanks for your help and your work.
Will test both, maybe it would make Sense to merge both.

closed 2020-03-04 by atze09

klacol, 2025-07-06:
> first

On Windows you can also make a connection test with Powershell, like this:

```shell
Test-NetConnection -ComputerName 192.168.0.10 -Port 502

ComputerName     : 192.168.0.10
RemoteAddress    : 192.168.0.10
RemotePort       : 502
InterfaceAlias   : WLAN
SourceAddress    : 192.168.0.63
TcpTestSucceeded : True
```

This is another way to verify the IP address and the port of the inverter (in my case it is 502 and not 1502).