binsentsu/home-assistant-solaredge-modbus #3: translations

Sparky32, 2020-05-19:
Hi . My hassio log gives a warning:
solaredge_modbus: the '.translations' directory has been moved, the new name is 'translations', starting with Home Assistant 0.111 your translations will no longer load if you do not move/rename this

Any id how to solve this ?

thavini, 2020-05-20:
You can easily fix it by renaming the folder in the components directory for now.
The fix needs to be done in the repo later on. 
For me in the folder: `/usr/share/hassio/homeassistant/custom_components/solaredge_modbus`

mv .translations translations

thavini, 2020-05-20:
Created a pull request also added Dutch translations. Maybe @binsentsu is able to find the fix suitable for his code and wil merge this.

For easy access in HACS also a new tag might be nice 😄.

Sparky32, 2020-05-20:
Thx for the quick reply .  I’l try that this weekend

binsentsu, 2020-05-22:
Hi thanks for reporting issue. PR #4 from @thavini merged with fix for translations. Release 1.1.0 just published. Closing this one.

closed 2020-05-22 by binsentsu