binsentsu/home-assistant-solaredge-modbus #5: New feature request: Multiple inverter units and meter

julezman, 2020-06-18:
Thanks for this great HACS component.  I'd like to request two new features:

1) Enhance the component so that it can read data from multiple inverters.
2) Add enhancement to read a Wattnode\SolarEdge modbus meter attached to the inverter.  This would primarily allow users to keep track of imported and exported power from the house as well as numerous other attributes the meter records.  

Didn't know if any work was going on in this regard and happy to help via a pull if not.

binsentsu, 2020-06-19:
Hi.
1. Reading data from multiple inverters should be possible. Just go through the integration UI configuration again for the second inverter.
2. I don't own a meter, so can't test this out myself. I think the fork of goergch /
home-assistant-solaredge-modbus includes meter implementation as far as I can see. Mayb you can check this out and test it. If it is ok, I can merge it.

miguelrjim, 2020-06-20:
I've tested `goergch` fork and it works exporting the meter information into HA; but the way the code is set up it will fail reading the registers if there's not a meter installed so I wouldn't recommend merging its fork as it is.

binsentsu, 2020-06-22:
@miguelrjim Thanks for the info. Are the meter and inverter hosted on the same IP address in such a setup?

stuarts0011, 2020-07-17:
> @miguelrjim Thanks for the info. Are the meter and inverter hosted on the same IP address in such a setup?

yes,

binsentsu, 2020-07-28:
There is currently a prerelease V1.2.0-A2 which has meter support. It would be great if it would be tested by some more people before release...

marpi82, 2020-07-30:
> There is currently a prerelease V1.2.0-A2 which has meter support. It would be great if it would be tested by some more people before release...

I am also testinh this release, I have one meter connected to the inverter - I have not noticed any problems so far

zeeuk, 2020-08-29:
Thank you very much! The added function works perfectly (meter)

binsentsu, 2020-09-26:
Finally released, closing this one...

closed 2020-09-26 by binsentsu