binsentsu/home-assistant-solaredge-modbus #6: Meter Device Block

vigne, 2020-07-15:
Hello,

thanks for the effort you put into this integration.

I was wondering, is there a setting to enable it also include the "Meter Device Block" defined in the SunSpec protocol?
My inverter can show also the energy exported/imported from the grid and it would be great to have this values in HA too.

I;d be happy to help (test, debug) if you're willing to cooperate.

Thanks!

binsentsu, 2020-07-18:
Hi. I just got a response on an earlier request concerning reading meter data as I don't own a meter. It seems this data is gathered from the same device (same IP). So it's fairly easy to incorporate this. Is the 'Meter 1' data enough? This starts from register 40190 until 40292

vigne, 2020-07-19:
Hi, thanks for responding. I'm not sure, but I think the groups 40226-40232 (M_Exported) and 40234-40240 (M_Imported) are the values I'm looking for. I assume they represent the watt-hours since the last polling, right?

binsentsu, 2020-07-24:
Hi,
I published a prerelease 1.2.0-A1 which contains support for Meter 1 data reading. Can you test it out? As it's a prerelease, I don't think it will be visible through hacs, so just copy paste to custom_components directory. Then re-add the integration and an extra config flow booleanentry 'Read Meter 1 Data' should be presented by the UI.

arundalep, 2020-07-26:
Thank you for your work on this. The inverter data is supurb and much faster than any other method.

I have tried the new version and selected M1 data but am nbot receiving it. It would be great to get M1 and M2 data using this if possible.

binsentsu, 2020-07-26:
@arundalep Thanks for testing. I updated the release (__init__.py) with an updated starting address. Can you retest? I assume you are seeing the new m1_ entities?

arundalep, 2020-07-26:
Just tried and it’s all working perfectly, thank you.
Would it be possible to add M2? 

Thank you so much

binsentsu, 2020-07-26:
I think it was a problem related to reading unsigned int instead of signed int. I fixed it in prerelease 1.2.0-A2. This one also adds support for meter 2 and 3. Just enable them in config UI. Thanks for giving feedback of your testing results!

arundalep, 2020-07-26:
So I have tested 1.2.0 A2, the AC Power now reads correctly (-ve signed for import +ve signed for export). The voltage, frequency and KWH meters all seem correct but the current and VA readings seem way high. (VA looks to be about double the actual reading). 
I haven’t had a chance to test M2 as it hasn’t arrived yet but will do later in the week. 

Thanks again for your efforts

vigne, 2020-07-27:
Hello, I too can conform seeing sensible data on meter 1. Thanks a lot, really appreciated!

binsentsu, 2020-07-27:
Hi. Just guessing: isn't a template sensor needed which takes both numbers of M1 and the solar panels into account? I don't have a solaredge meter myself, but I'm using a shelly PM with CT clamp and have such a calculated sensor for total power consumption. eg:
      energy_calculated_consumption:
        friendly_name: 'Energy Calculated Consumption'
        entity_id:
          - sensor.shelly_em_b9e8cc_meter_power_0
          - sensor.solaredge_ac_power
        value_template: "{{ (states('sensor.shelly_em_b9e8cc_meter_power_0')|float + states('sensor.solaredge_ac_power')|float)|round(3) }}"
        unit_of_measurement: "W"

binsentsu, 2020-07-27:
@arundalep Do you still have any doubts about the numbers? I double checked and the implementation should now be coherent with the specification.

vigne, 2020-07-27:
Hi, 

I agree, a template sensor will work too, I just thought this value is in the the protocol. 
But anyway, thanks a lot for this integration and also for extending it with meter data.

Cheers!

binsentsu, 2020-09-26:
Finally released, closing this one.

closed 2020-09-26 by binsentsu