binsentsu/home-assistant-solaredge-modbus #8: New feature request: Ability to setup by config/flow modbus device unit

marpi82, 2020-07-28:
Hello,

Most of inverters are running as first (unit 0x01) on modbus (this is Yours default also)
My SE inverter are running as second device (unit 0x02) and I've must change two valuse in __init__.py from 1 to 2 to get it work in my case.
It will be nice if we can setup unit number from config/flow

Marek

binsentsu, 2020-07-28:
To be sure, do you mean this one: self.read_holding_registers(**unit=1**, address=40071, count=38)

marpi82, 2020-07-29:
> 
> 
> To be sure, do you mean this one: self.read_holding_registers(**unit=1**, address=40071, count=38)

Yes, this one on master repository:
https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/master/custom_components/solaredge_modbus/__init__.py#L182

And this two lines on feature/meter1 (I have used this repository):
https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/feature/meter1/custom_components/solaredge_modbus/__init__.py#L338
https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/feature/meter1/custom_components/solaredge_modbus/__init__.py#L632

Marek

johnpwhite, 2021-05-16:
Hi,

I'd love this feature too please! I too have to change these two values to equal 2 to match my meter device ID of 2.

THANKS IN ADVANCE!!

marpi82, 2021-08-06:
Is there a chance that this parameter will appear in the configuration? :)

binsentsu, 2022-10-22:
implemented

closed 2022-10-22 by binsentsu