binsentsu/home-assistant-solaredge-modbus #10: Feature Request: Configure Button

alexborro, 2020-09-28:
After adding the integration, there is no Configure/Option button to change any parameter, like the IP of the inverter.
One should to remove the integration and add again to change any param.

mirceadamian, 2020-11-16:
Funny enough, I have ran into this issue myself (changing the IP address).
Is there any way to manually change the IP address?
I could not figure out in which file it is stored.

alexborro, 2020-11-16:
Nope.. you need to remove and configure it again.

mirceadamian, 2020-11-16:
It has to be stored somewhere.
I have found a trace of it in the file _.storage/core.config_entries_ locate under the config directory.
```
...
            {
                "entry_id": "9f52af74927c00286355c60e566ad371",
                "version": 1,
                "domain": "solaredge_modbus",
                "title": "solaredge",
                "data": {
                    "name": "solaredge",
                    "port": 1502,
                    "read_meter_1": false,
                    "read_meter_2": false,
                    "read_meter_3": false,
                    "scan_interval": 10,
                    "host": "192.168.178.53"
                },
                "options": {},
                "system_options": {
                    "disable_new_entities": false
                },
                "source": "user",
                "connection_class": "local_poll",
                "unique_id": "192.168.178.53"
            }
...
```

ChristophCaina, 2022-11-02:
reconfiguration of the integration documented in the discussions area.
- this could probably be closed :)

closed 2022-11-09 by binsentsu