binsentsu/home-assistant-solaredge-modbus #14: no stats while connected

kxqcvl, 2020-10-26:
Hey there, i'm new to home assistent and love this integration, thanks for that!
The issue i'm facing is that i can connect to my inverter (display @ tcp says connected when i start integration) but i can't seem to get any data from my entities, they al say "unknow". I've restarted HA several times (inverter says "ready" when i restart HA en back to "connected" when HA has rebooted) but still no luck. I'm 99,9% shure i'm connected but no data is coming true.
Any idea?

thanks.

binsentsu, 2020-10-28:
This integration reads from the default device ID (1). Mayb the modbus device ID on your inverter is different?

kxqcvl, 2020-10-29:
> 
> 
> This integration reads from the default device ID (1). Mayb the modbus device ID on your inverter is different?

Is there any way i can change it in the configuration files so i can test it?
I can't seem to find any reference @ device ID.

thanks.

binsentsu, 2020-10-29:
At the moment the integration only supports default device id. You van check issue #8 for local workaround for testing. Alsof check the solaredge specs that your device is certainly supported ( https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf )

kxqcvl, 2020-10-31:
thanks, i've tried to change to device id and set it to 2 but doesn't seems to be the solution for me.
I've looked at the logs but don't see anything that the integration couldn't retreive information.
Is there any way that i can check what the problem is?

thanks

closed 2022-10-22 by binsentsu