binsentsu/home-assistant-solaredge-modbus #17: Not working if I'm connecting to a meter

robi59italy, 2021-03-16:
I have a Solaredge SE4000H, connected by ethernet and Rasberry with home assistant, updated to last version.

If I'm installing this integration by hacs and without meter, it's working.

If I'm trying to select a meter (1 or 2 or 3) I'm getting "unknown" value.

My meter is connected by RS485 and it's the standard modbus meter.

Even I have a battery connected to the inverter, and I can't see the level but this is another history...

robi59italy, 2021-03-17:
Update
I make a new configuration with the FV running.. I can see the value if after highlighted meter 1 but all the values are zero.

and the AC Power is showing wrong data's... in the night is showing the real home consumption... now with FV running it's showing its production :-(

binsentsu, 2021-03-20:
Hi, I don't have meters or battery myself at home. The data is directly read from the registers according to the sunspec, don't know what the influence of the FV is. Trying for example this library manually (https://github.com/nmakel/solaredge_modbus/) to see if it shows the same results, could give you a direction.

robi59italy, 2021-03-20:
> Hi, I don't have meters or battery myself at home. The data is directly read from the registers according to the sunspec, don't know what the influence of the FV is. Trying for example this library [...]

Thanks for yr support.
I checked witht the suggested python library and it's showing the same numbers I can see in HA, including a long list of zero from smartmeter and battery.

Now I contacted Solaredge to ask support and see if I have to update the firmware. Maybe this could help

TheLordVader, 2021-04-06:
How did it go? I have modbus meter via RS485 as well.

robi59italy, 2021-04-06:
> How did it go? I have modbus meter via RS485 as well.

Solaredge updated my inverter remotely. I'm getting same data's but not all. The reason is Solaredge is applying Sunspec specification and this spec doesn't include all informations .. :-(

TheLordVader, 2021-04-06:
Did you try this?

- Flick the switch on SolarEdge to P and connect your mobile device to the WiFi hotspot (password on the side of the inverter, or use the QR code + mySolarEdge app)

- Open the browser and visit http://172.16.0.1 41 and click on the “Communication” menu item
Select “RS485-2”

- Choose Protocol “SunSpec (Non-SE Logger)”, also under that set Device ID to 1 (mine was set to 5). Now some Modbus TCP requests were responding.

- Change the Protocol over to “SolarEdge Master” (now a SolarEdge logger script was getting data too)

- Change back protocol to “None”

- Power cycle the inverter for good measure, and it was all still working

Let me know if that helps.

robi59italy, 2021-04-06:
thanks I will do. My inverter is y2018 and it has the LCD panel but the procedure should be the same.

stepir, 2021-04-06:
I have 2020 model (no display) and browsing its IP don't expose a web UI. However I was able to enable modbus with the SetApp app. It works w/o meters but most of the useful information are missing (production, consumption and batteries levels). With the meters enabled everything stay unknown:

[image]

[image]

I have 3 inverters, 1 is the production and the 2 others are 5k connected to LGCHem batteries - one those 2 is the master (the one where I have enabled modbus)

robi59italy, 2021-04-06:
> Did you try this?
> 
> * Flick the switch on SolarEdge to P and connect your mobile device to the WiFi hotspot (password on the side of the inverter, or use the QR code + mySolarEdge app)
> * Open the browser and visit http://172.16.0.141 and click on the “Communication” menu item
>   Select “RS485-2”
> * Choose Protocol “SunSpec (Non-SE Logger)”, also under that set Device ID to 1 (mine was set to 5). Now some Modbus TCP requests were responding.
> * Change the Protocol over to “SolarEdge Master” (now a SolarEdge logger script was getting data too)
> * Change back protocol to “None”
> * Power cycle the inverter for good measure, and it was all still working
> 
> Let me know if that helps.

Tried this and it didn't work. I'm interesting to see the screenshot of somebody working with modbus + battery + smartmeter by solaredge to see which kind of data's is getting

stepir, 2021-04-06:
> Tried this and it didn't work. I'm interesting to see the screenshot of somebody working with modbus + battery + smartmeter by solaredge to see which kind of data's is getting

@robi59italy see above - I don't get data but haven't switch to SunSpec protocol as my master inverter won't answer to http requests...

TheLordVader, 2021-04-06:
Too bad..! 
I don't have a battery (SE15k only), but reading through the thread from this hass addon [https://github.com/erikarenhill/solaredge-modbus-hass] it seemes someone got it working with Smart Meter. Don't know the difference between the versions, and I really don't know which one to try, I have HACS also.

stepir, 2021-04-09:
so tried and I don't think I can do it - I have 3 inverters (1 for handling 1 battery and acting as master, 1 for handling a second battery and acting as slave, 1 for the production and acting as slave). They are connected in RS485 on port 2. I think that such change will break the site communications between my inverters:

[image]

[image]

[image]

purcell-lab, 2021-04-16:
I have two inverters and one meter which appears to be working if I select MODBUS and only meter 1, if I select any other meter I get 'unknown' for all values.

Some screen shots:

[image]

[image]

[image]

[image]

[image]

stepir, 2021-04-16:
Yes as said I have a master (with battery and meter) + 2 slave (1 for PV and 1 for another master). I see the same results.

BTW the thing I'm mostly missing is then the PV and the overall battery capacity and power of my batteries.

The latter can be achieved by querying the following registers

- name: battery
  type: default
  power:
    type: modbus
    uri: 192.168.2.15:502
    id: 1
    register:
      address: 62836
      type: holding
      decode: float32s
    scale: -1
  soc:
    type: modbus
    uri: 192.168.2.15:502
    id: 1
    register:
      address: 62852
      type: holding
      decode: float32s

This is my understanding by using https://github.com/andig/evcc and haven't still figure out the PV part.

The end story is that if this plugin won't come up with such info I don't see how we could replace the standard cloud based plugin of HA.

robi59italy, 2021-04-16:
@stepir you are right, I'm working with a mix .. the hope is sunspec protocol will include FV info in the future.

stepir, 2021-04-16:
check this thread at evcc as they seems to have figure out also the FV

https://github.com/andig/evcc/discussions/665

robi59italy, 2021-04-16:
> check this thread at evcc as they seems to have figure out also the FV
> 
> [andig/evcc#665](https://github.com/andig/evcc/discussions/665)

Yes very good thanks

TheLordVader, 2021-10-12:
Yes, that's correct. Please post how it went :-)

TheLordVader, 2021-10-12:
> sorry - another question:
>
> If I have the following configuration on the Inverter:
>
> RS485-1 -> Modbus device -> Meter1 -> Import / Export
>
> should I use Meter2 for RS485-2 ?
> Or should I also use Meter1?
> As far as I can remember, this caused the issue, that Meter1 -> Import /
> Export was being removed from the RS485-1 configuration...

Well, there seems to be some issue with some settings remaining remembered
in the add on. I would configure the inverter to talk with the new meter on
channel 2, that way it won't cause any issues on bisentsu add on.

But, I you are willing to try meter 1 again I personally would go for that
first.

ChristophCaina, 2021-10-20:
I would suggest to add a setup guide to the Wiki - and this issue itself could be closed then... ?

pczolee, 2022-01-11:
Hello.

I have the same problem, If I setup the meter 1 then there is no data. In the log there is some error:

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 229, in read_modbus_data
    and self.read_modbus_data_meter1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 241, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 405, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

Without meter 1 everything looks good. I tried to change the protocol to sunspec, but if i do it than I cannot use the meter to limit my export to grid energy... So the question is how can I manage to work the energy meter with inverter and to read data from it.

Thanks

mpredfearn, 2022-01-11:
@pczolee the value being read from the meter is failing validation:
`
exported = validate(self.calculate_value(exported, energywsf), ">", 0)
File "/config/custom_components/solaredge_modbus/init.py", line 124, in validate
raise ValueError(f"Value {value} failed validation ({comparison}{against})")
**ValueError: Value 0 failed validation (>0)**
`
The meter value for "exported energy" is 0 - that is likely impossible as it implies that the system has never exported any power. Are you sure that your system is physically wired / configured to use meter 1?

pczolee, 2022-01-12:
> @pczolee the value being read from the meter is failing validation:
>
> exported = validate(self.calculate_value(exported, energywsf), ">", 0) File "/config/custom_components/solaredge_modbus/init.py", [...]

Well, I use the meter to limit the export. So the export is set to 0W. So the answer is, it can be 0. Otherwise the wiring etc is good, because I can see the values with setapp...
Thanks

mpredfearn, 2022-01-12:
@pczolee wow I'm amazed that even with export limit set to 0, 0Wh of energy have been exported in all time - I would have expected a small amount of leakage as the system reacts to transients, etc.

Anyway, I've made a PR to remove the validation of export being > 0. The only other validation performed of meter registers is that import is > 0. I assume that this will always be the case because surely all systems must pull from the grid at some point.

https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/76

pczolee, 2022-01-12:
> @pczolee wow I'm amazed that even with export limit set to 0, 0Wh of energy have been exported in all time - I would have expected a small amount of leakage as the system reacts to transients, etc. [...]

Thanks. I going to have a look and let you know!

pczolee, 2022-01-12:
> @pczolee wow I'm amazed that even with export limit set to 0, 0Wh of energy have been exported in all time - I would have expected a small amount of leakage as the system reacts to transients, etc. [...]

Unfortunately, still there is something wrong:

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 229, in read_modbus_data
    and self.read_modbus_data_meter1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 241, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 409, in read_modbus_data_meter
    imported = validate(self.calculate_value(imported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

mpredfearn, 2022-01-12:
Hey @pczolee this is the validation of the *imported* energy failing - it is also zero. Your system has never measured any imported energy either?

pczolee, 2022-01-12:
@mpredfearn Well, of course the imported is not zero! What else can be the problem?

pczolee, 2022-01-12:
Well, with https://github.com/nmakel/solaredge_modbus/ I try to connect (with example.py), and it show:

Registers:
	Manufacturer: SolarEdge
	Model: SE50K-RW0P0BNY4
	Type: Three Phase Inverter
	Version: 0004.0014.0107
	Serial: XXXXXXX
	Status: Producing
	Temperature: 39.12°C
	Current: 13.06A
	Phase 1 Current: 4.31A
	Phase 2 Current: 4.40A
	Phase 3 Current: 4.34A
	Phase 1 voltage: 404.10V
	Phase 2 voltage: 405.80V
	Phase 3 voltage: 406.20V
	Phase 1-N voltage: 234.00V
	Phase 2-N voltage: 232.80V
	Phase 3-N voltage: 235.40V
	Frequency: 50.03Hz
	Power: 2919.00W
	Power (Apparent): 3033.00VA
	Power (Reactive): -826.00VAr
	Power Factor: -98.42%
	Total Energy: 38289Wh
	DC Current: 3.97A
	DC Voltage: 0.00V
	DC Power: 2963.40W

Should I see the meter as well here?

pczolee, 2022-01-12:
I think I have to give the unit for the example.py, so the meter have number 2 address:

./example.py --unit 2 192.168.8.7 1502
Inverter(192.168.8.7:1502, connectionType.TCP: timeout=1, retries=3, unit=0x2):

Registers:
Traceback (most recent call last):
  File "./example.py", line 46, in <module>
    print(f"\tManufacturer: {values['c_manufacturer']}")
KeyError: 'c_manufacturer'

If I right than I should see the meter details here, but nothing... I have no idea why. If it helping I can make screenshot from the setapp.

mpredfearn, 2022-01-12:
@pczolee The register addresses of the meters are different to the inverter, so just changing the unit address won't work. The inverter registers are at address 40071, meter 1 at 40190, meter 2 at 40364 and meter 3 at 40539 all on the same unit address.

I haven't used https://github.com/nmakel/solaredge_modbus/ but looks like a useful library. The example.py doesn't appear to support reading meters, but there is documentation in the Readme about how to read them from an interactive Python shell. It would definitely be worth trying to read the meter registers that way to see if it agrees with what the integration is reading.

pczolee, 2022-01-12:
Are there any tool what I can use to test that the communication via modbus tcp with the meter is working? The https://github.com/nmakel/solaredge_modbus/ can read the meter I just don't know how should I use to do it...

mpredfearn, 2022-01-12:
Hey @pczolee - so I had a go with the library - example.py does support reading meters / batteries but it only dumps their content to json. FYI here is what I get if I run it for my system:
`
./example.py --json 192.168.1.187 502 
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE3680",
    "c_version": "0003.2724",
    "c_serialnumber": "731C26F1",
    "c_deviceaddress": 1,
    "c_sunspec_did": 101,
    "current": 167,
    "l1_current": 167,
    "l2_current": 0,
    "l3_current": 0,
    "current_scale": -2,
    "l1_voltage": 2363,
    "l2_voltage": 0,
    "l3_voltage": 0,
    "l1n_voltage": 0,
    "l2n_voltage": 0,
    "l3n_voltage": 0,
    "voltage_scale": -1,
    "power_ac": 3649,
    "power_ac_scale": -1,
    "frequency": 50026,
    "frequency_scale": -3,
    "power_apparent": 3966,
    "power_apparent_scale": -1,
    "power_reactive": 15543,
    "power_reactive_scale": -2,
    "power_factor": 9200,
    "power_factor_scale": -2,
    "energy_total": 12245720,
    "energy_total_scale": 0,
    "current_dc": 8914,
    "current_dc_scale": -4,
    "voltage_dc": 4155,
    "voltage_dc_scale": -1,
    "power_dc": 3704,
    "power_dc_scale": -1,
    "temperature": 3840,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "WattNode",
            "c_model": "WND-3Y-400-MB",
            "c_option": "Export+Import",
            "c_version": "25",
            "c_serialnumber": "4125226",
            "c_deviceaddress": 2,
            "c_sunspec_did": 203,
            "current": 0,
            "l1_current": 0,
            "l2_current": 0,
            "l3_current": 0,
            "current_scale": -1,
            "voltage_ln": 23652,
            "l1n_voltage": 23652,
            "l2n_voltage": 0,
            "l3n_voltage": 0,
            "voltage_ll": 23446,
            "l12_voltage": 23430,
            "l23_voltage": 0,
            "l31_voltage": 23462,
            "voltage_scale": -2,
[... 96 lines cut ...]
            "maximum_charge_continuous_power": -3.4028234663852886e+38,
            "maximum_discharge_continuous_power": -3.4028234663852886e+38,
            "maximum_charge_peak_power": -3.4028234663852886e+38,
            "maximum_discharge_peak_power": -3.4028234663852886e+38,
            "average_temperature": -3.4028234663852886e+38,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": -3.4028234663852886e+38,
            "instantaneous_current": -3.4028234663852886e+38,
            "instantaneous_power": 0.0,
            "lifetime_export_energy_counter": 0,
            "lifetime_import_energy_counter": 0,
            "maximum_energy": -3.4028234663852886e+38,
            "available_energy": -3.4028234663852886e+38,
            "soh": -3.4028234663852886e+38,
            "soe": -3.4028234663852886e+38,
            "status": 7,
            "status_internal": 0,
            "event_log": 0,
            "event_log_internal": 0
        }
    }
}
`
You can see that my meter 1 is configured as an export / import meter and it's export / import values are 4.2MWh / 7.8MWh respectively. That lines up with what the integration is reading:
[image]

pczolee, 2022-01-12:
Thanks. So i tried it:
example.py --json 192.168.8.7 1502
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE50K-RW0P0BNY4",
    "c_version": "0004.0014.0107",
    "c_serialnumber": "XXXXXX",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "current": 1469,
    "l1_current": 492,
    "l2_current": 489,
    "l3_current": 487,
    "current_scale": -2,
    "l1_voltage": 4050,
    "l2_voltage": 4065,
    "l3_voltage": 4076,
    "l1n_voltage": 2346,
    "l2n_voltage": 2330,
    "l3n_voltage": 2361,
    "voltage_scale": -1,
    "power_ac": 3329,
    "power_ac_scale": 0,
    "frequency": 5004,
    "frequency_scale": -2,
    "power_apparent": 3435,
    "power_apparent_scale": 0,
    "power_reactive": -8480,
    "power_reactive_scale": -1,
    "power_factor": -9846,
    "power_factor_scale": -2,
    "energy_total": 40045,
    "energy_total_scale": 0,
    "current_dc": 45216,
    "current_dc_scale": -4,
    "voltage_dc": 0,
    "voltage_dc_scale": -32768,
    "power_dc": 3379,
    "power_dc_scale": 0,
    "temperature": 4018,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "meters": {},
    "batteries": {}
}

As you can see no meters there :( . I'm going to have some screenshot from the setapp, maybe it help to find the problem...

mpredfearn, 2022-01-12:
@pczolee OK so that means that the Meter DID registers 0x9cfc, 0x9daa, 0x9e59 are all 0. Seems strange that the inverter would be reporting these as 0 even if it is configured to use a meter. Very odd.

pczolee, 2022-01-12:
So my setup is:
[image]
[image]
[image]

The meter is working:
[image]

This is the other RS485 port setting (this is empty)
[image]

The power control part:
[image]
[image]
[image]

And the info page:
[image]

I have no idea what is the problem :(

mpredfearn, 2022-01-12:
Hey @pczolee Yeah I have no idea - everything looks correct in SetApp and yet the modbus registers are not reflecting the set up correctly. To be honest, you might have to go to Solaredge support with this one, sorry.

Note that even though your export limit is set to 0, 12.45kWh has been exported - so as I expected some has leaked due to transients etc. That means the validation we do on the meter values is OK - since the value that *should* be read from the register is not 0. 

Sorry I'm not sure I can help more.

pepsonEL, 2022-01-12:
You are conecting by wifi or by LAN ? In new FW solaredge comunication by modbus on wifi was blocked and works only LAN.,

pczolee, 2022-01-12:
@mpredfearn Yes, I'm going to talk with the support! Well, about the 12.45kWh, there was a few hours when exported (before I set up the limit).

@pepsonEL We are on LAN. But thanks for the info ;)

pirpy, 2022-04-12:
Hello
I recently add meter 2 to the system  but if I add meter 1 and 2 then no data is received
it only works with meter 1.
Any sugestions

stepir, 2022-04-13:
> Hello
> I recently add meter 2 to the system but if I add meter 1 and 2 then no data is received
> it only works with meter 1.
> Any sugestions

exactly my same experience

lorenzing, 2022-10-23:
Hi @pczolee, have you solve the problem? I'm in the same situation. Thank you so much!

pczolee, 2022-10-24:
> Hi @pczolee, have you solve the problem? I'm in the same situation. Thank you so much!

Hi.

No, unfortunately not. I wrote to the SE support, they promised that they will sort it out and let me know, but there was no reply...

stepir, 2022-10-24:
it's most probably a modbus misconfiguration - you might want also to check the SolarEdge Modbus Multi Device integration which seems to be better maintained https://github.com/WillCodeForCats/solaredge-modbus-multi