binsentsu/home-assistant-solaredge-modbus #18: Feature Request: Get "Lifetime Data"

ChristophCaina, 2021-04-03:
HI, is it possible to get some sort of Lifetime Data like the kWh production per Hour, Day, Week, Month, year and Total?
Or how can this be acheived with the current version of this integration?

Thanks
And with best regards,
Christoph

purcell-lab, 2021-04-16:
I seem to have lifetime data reported as (AC energy KWH), you could use 'utility meter integration' to get day/ month/ year. 

[image]

WillCodeForCats, 2021-09-13:
Now that Home Assistant has the Energy dashboard, just set that up.

binsentsu, 2021-09-15:
Closing this one. As options of Utility Meter and Energy dashboard are resolving it.

closed 2021-09-15 by binsentsu