binsentsu/home-assistant-solaredge-modbus #25: Help with the Integration

Subia75, 2021-08-10:
I need help installing the integration. 

I have a I install through HACS but after reboot of Home-Assistant, I don't see the integration then i can't be configured it
I have hassio  2021.8.5 version.
I don't understand what I'm doing wrong. 

I ask if possible help, I do not think it is a problem related to the 2021. 8release

Thank

grouts112, 2021-08-11:
Once installed via HACS,  did you then go to Home assistant Configuration —> Integrations
And added the integration ?
HACS just installs it to HA,  you then need to add the integration.

Subia75, 2021-08-11:
Thanks for the reply.
I solved
I didn't understand why I couldn't find the integration after a reboot.
Then I turned off the power to the raspberry and on reboot it included me in the integrations.
Sorry I could have tried before asking.

closed 2021-08-11 by Subia75