binsentsu/home-assistant-solaredge-modbus #31: Sensor AC Energy kWh not compatible with new Energy feature

bekesizl, 2021-09-05:
Sensor "sensor.solaredge_ac_energy_kwh (measurement)" cannot be configured as Solar production source in new Energy feature.
Error message at configuration says: entity doesn't have state_class "total_increasing".

Would be nice if this integration could also be used as a source for solar production, not only the official integraion.

thavini, 2021-09-05:
I have now the following and it works for me.
It allows me toe set up the dashboard and t9 me the values are all correct. 

[image]

bekesizl, 2021-09-05:
I am a novice in Home Assistant. Where did you change this?

I use the HACS integration version and in Development tools / States I see following:

state_class: measurement
last_reset: 1970-01-01T00:00:00+00:00
unit_of_measurement: kWh
friendly_name: solaredge (AC Energy KWH)
icon: mdi:solar-power
device_class: energy

thavini, 2021-09-05:
I have this in my customize.yml.

But your configuration might be different but still you should be able to override the state_class. That is basically what's needed to be changed here. Change measurement into total_increasing.

bekesizl, 2021-09-05:
Thank you. I managed to get it running.

sebnaf, 2021-09-05:
> Thank you. I managed to get it running.

Can you post your config for reference? Thanks

bekesizl, 2021-09-06:
**configuration.yaml:**
```
homeassistant:
  customize: !include customize.yaml
```
**customize.yaml:**
```
sensor.solaredge_ac_energy_kwh:
  device_class: energy
  state_class: total_increasing
  friendly_name: "Solaregde AC energy kWh"
```

binsentsu, 2021-09-07:
Version 1.3.1 released with fix to be compatible with HA 2021.9

closed 2021-09-07 by binsentsu