binsentsu/home-assistant-solaredge-modbus #34: KWh wrong values on my installation (?)

ctrl-alt-d, 2021-09-07:
Hi, thanks about this great integration, nice software!

After install this great integration, I can see metrics that are accurate with consumption. For example, solar production:

[image]

Also, I can see how many kw I'm getting from grid (-), or I'm sending to grid (+). In this case, I'm getting 130W from grid:

[image]

But I can not use this metrics when I configure HA Energy (Maybe because they are on W and not on KWh).

My issue is with the KWh metrics:

Solar production KWh looks wrong:

[image]

Meter, import KWh looks wrong:

[image]

Meter, output KWh looks wrong:

[image]

Then, the dashboard has no sense:

[image]

I have opened this issue because I don't know what I'm doing wrong. Maybe someone can help me with HA energy configuration.

Thanks a lot, and sorry about my noob question.

mpredfearn, 2021-09-08:
The Meter kWh values are cumulative since the meter was installed. So over the lifetime of your system, you have exported ~567kWh and imported ~155kWh. I can't see "AC Energy" in your screenshots but it is also the cumulative energy your system has produced, so based on the Solaredge portal I would expect that sensor to have a value around 1.17MWh.
You need to look at how much the meter readings change over time (which is what the HA dashboard is doing) to determine the energy flow.
You have your energy dashboard set up exactly as I have, with the same sensors. On the Solaredge portal, you need to look at the energy graphs below the power flows (power is instantaneous, energy is cumulative) and compare it to HAs idea of the energy distribution.

ctrl-alt-d, 2021-09-09:
thanks @mpredfearn

closed 2021-09-09 by ctrl-alt-d