binsentsu/home-assistant-solaredge-modbus #36: Setting battery charge/discharge profile

nerfherder, 2021-09-08:
It is possible to control battery charge and discharge profiles by setting values to modbus register 63242 provided register 63236 is set to remote control.

I have set this up using the default modbus integration in hassio but this would be better setup as a default control in this integration.

The registers and settings are defined in https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/

mpredfearn, 2021-09-09:
Nice - I haven't seen the Solaredge documentation but that should help me fine tune my PR #35 - although my LG battery does not seem to exactly fit the documented register behaviours.

It would be awesome to add support for changing the storage mode - we probably just have to add a service that can be called to switch it by writing the appropriate registers. I'm certainly keen to do this myself as I have a problem right now with the inverter dumping the battery charge into the EV when it starts charging during hours of cheap import. If I could have HA switch the profile automatically, then I can hold on to the battery charge for when the import rate is expensive again.

nerfherder, 2021-09-10:
Yeah it's very handy to be able to change.
Currently I just write to register 63242 in an automation based on a input_select field.

My automation yaml is below

```
alias: Battery Profile
description: Change battery profile from UI
trigger:
  - platform: state
    entity_id: input_select.battery_profile
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_select.battery_profile
            state: 'Off'
          - condition: not
            conditions:
              - condition: state
                entity_id: sensor.battery_profile
                state: '0'
        sequence:
          - service: modbus.write_register
            data:
              address: 63242
              unit: 1
              value: 0
              hub: solaredge
      - conditions:
          - condition: state
            entity_id: input_select.battery_profile
            state: Charge from excess PV
          - condition: not
            conditions:
              - condition: state
                entity_id: sensor.battery_profile
                state: '1'
        sequence:
          - service: modbus.write_register
            data:
              address: 63242
              unit: 1
              value: 1
              hub: solaredge
      - conditions:
          - condition: state
            entity_id: input_select.battery_profile
            state: Charge from PV first
          - condition: not
            conditions:
              - condition: state
                entity_id: sensor.battery_profile
                state: '2'
        sequence:
          - service: modbus.write_register
            data:
              address: 63242
              unit: 1
              value: 2
              hub: solaredge
      - conditions:
          - condition: state
            entity_id: input_select.battery_profile
            state: Charge from PV and Grid
          - condition: not
            conditions:
              - condition: state
                entity_id: sensor.battery_profile
                state: '3'
        sequence:
          - service: modbus.write_register
            data:
              address: 63242
              unit: 1
              value: 3
              hub: solaredge
      - conditions:
          - condition: state
            entity_id: input_select.battery_profile
            state: Maximise export
          - condition: not
            conditions:
              - condition: state
                entity_id: sensor.battery_profile
                state: '4'
        sequence:
          - service: modbus.write_register
            data:
              address: 63242
              unit: 1
              value: 4
              hub: solaredge
      - conditions:
          - condition: state
            entity_id: input_select.battery_profile
            state: Maximise Self-Consumption
          - condition: not
            conditions:
              - condition: state
                entity_id: sensor.battery_profile
                state: '7'
        sequence:
          - service: modbus.write_register
            data:
              address: 63242
              unit: 1
              value: 7
              hub: solaredge
    default: []
mode: single
```

mpredfearn, 2021-09-14:
@nerfherder Thanks for the link to the documentation! I've used it to implement control of my battery's charge / discharge profile :-) I now have control!

mpredfearn, 2021-09-21:
Implemented in #38

closed 2021-09-21 by nerfherder