binsentsu/home-assistant-solaredge-modbus #37: Data errors with fast refresh.

nerfherder, 2021-09-14:
I have been facing some issues with nonsense data coming out of this integration on and off.

This wasn't an issue with https://github.com/erikarenhill/solaredge-modbus-hass and having a look their it looks like erikarenhill is using import pymodbusTCP whereas bisentsu is using pymodbus the pymodbusTCP implementation lets you keep the connection open with `ModbusClient(host, port=port, unit_id=1, auto_open=True)` whereas pymodbus appears to require a connect and close each read cycle.

My suspicion is that this open/close cycle can take longer than 5-10 seconds and cause errors if it hasn't run through properly. I would like to trial using pymodbusTCP to see if it makes any difference.

_Originally posted by @nerfherder in https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/35#issuecomment-918691933_

WillCodeForCats, 2021-10-18:
Does HA actually allow for holding the socket open between polling cycles? I don't really know the internals of how HA add ons work.

From my experience with libmodbus in C, ideally the socket would be opened when HA starts up and then never closed unless 1) connection drops and needs to be reopened (network failure, inverter loses power and reboots, whatever) or 2) HA shuts down.

WillCodeForCats, 2021-10-19:
Wireshark would confirm if it's never intentionally closing the connection if someone has time to try that.

nerfherder, 2021-10-19:
I will have a look into monitoring with wireshark as I am already running on that fork.