binsentsu/home-assistant-solaredge-modbus #40: modbus integration not found in HA

RemcoLeonard, 2021-09-16:
I am trying to install the modbus integration for my SolarEdge. 
After installation of the component via HACS and restart of HA modules, i cannot find the integration in the list. Only the 'official' integration. 
tried to remove the component and restart of HA, and manually checked the filesystem to clear remains of the component and rein stalling it, but no luck so far. 
Where can i look further?
Thanks!

binsentsu, 2021-09-16:
First check if it's available in your custom component folder. If so it could be a caching problem of the UI layer in the browser which prevents showing the integration in the list. Try hard refreshing the webpage. 
Windows: ctrl + F5
Mac/Apple: Apple + R of command + R
Linux: F5

RemcoLeonard, 2021-09-20:
> First check if it's available in your custom component folder. If so it could be a caching problem of the UI layer in the browser which prevents showing the integration in the list. Try hard refreshing the webpage.
> Windows: ctrl + F5
> Mac/Apple: Apple + R of command + R
> Linux: F5

Thanks! This has fixed my issue. I'll close this..

closed 2021-09-20 by RemcoLeonard