binsentsu/home-assistant-solaredge-modbus #41: Direct usage of energy stats are incompatible with Home Assistant energy dashboard

mpredfearn, 2021-09-20:
Continuing the discussion from https://github.com/home-assistant/core/issues/55974 with @emontnemery.

When used with a DC coupled battery, the metrics reported by this integration as direct reads of the inverter sensors are currently incompatible with the Home assistant energy dashboard.
The "AC Energy kWh" sensor reports the lifeltime generated energy, but if this is used as the "solar production" sensor for the energy dashboard, it does not include the energy charged to the battery, as assumed / required by HA. It does include the energy discharged from the battery, which is assumed / required not to be the case by HA.

Another sensor is required for "Solar production" which is calculated as:
"AC Energy kWh" + "Battery charged" - "Battery discharged".

An additional problem is that, on my LG RESU 7kWh HV battery at least, the "battery charged" and "battery discharged" sensors reset at midnight (in violation of the documented protocol which states that they are lifetime sensors). This issue of the energy sensors resetting needs to be addressed in the integration either by fixing the existing ones to turn them into lifetime sensors, or add additional sensors that really are lifetime sensors.
An alternative might be to use an "integration" integration to convert battery power into battery energy, but that requires creating additional templated sensors which is a bit of a PITA.

ministryofsillywalks, 2021-10-01:
Are you working on changing this integration or did you end up making a new template sensor?
Just got my pv with a battery installed and am facing the same issue.

mpredfearn, 2021-10-04:
@ministryofsillywalks for now, I ended up using templated sensors, but I am thinking about adding some calculated sensors to the integration itself. I've got so many PRs open I'm waiting for movement on them before working on more. FYI here are my templated sensors for now.
```
      power_battery_charging:                                                   
        friendly_name: "Power - Battery Charging"                               
        unit_of_measurement: "W"                                                
        value_template: "{{ states('sensor.solaredge_modbus_battery1_power') | float | max(0) | abs() }}"
      power_battery_discharging:                                                
        friendly_name: "Power - Battery Discharging"                            
        unit_of_measurement: "W"                                                
        value_template: "{{ states('sensor.solaredge_modbus_battery1_power') | float | min(0) | abs() }}"
      power_grid_import:                                                        
        friendly_name: "Power - Grid Import"                                    
        unit_of_measurement: "W"                                                
        value_template: "{{ states('sensor.solaredge_modbus_m1_ac_power') | float | min(0) | abs() }}"
      power_grid_export:                                                        
        friendly_name: "Power - Grid Export"                                    
        unit_of_measurement: "W"                                                
        value_template: "{{ states('sensor.solaredge_modbus_m1_ac_power') | float | max(0) | abs() }}"
      power_solar_generation:                                                   
        friendly_name: "Power - Solar Generation"                               
        unit_of_measurement: "W"                                                
        value_template: "{{ (states('sensor.solaredge_modbus_i1_ac_power') | float) + (states('sensor.solaredge_modbus_battery1_power') | float) }}"
      power_consumption:                                                        
        friendly_name: "Power - Consumption"                                    
        unit_of_measurement: "W"                                                
        value_template: "{{ (states('sensor.solaredge_modbus_i1_ac_power') | float) - (states('sensor.solaredge_modbus_m1_ac_power') | float) }}"
      power_self_consumption:                                                   
        friendly_name: "Power - Self Consumption"                               
        unit_of_measurement: "W"                                                
        value_template: "{{ (states('sensor.solaredge_modbus_i1_ac_power') | float) - (states('sensor.power_battery_discharging') | float) - states('sensor.power_grid_export') | float}}"
                                                                                
      energy_solar_generation:                                                  
        friendly_name: "Energy - Solar Generation"                              
        unit_of_measurement: "kWh"                                              
        value_template: "{{ (states('sensor.solaredge_modbus_i1_ac_energy_kwh') | float) - (states('sensor.energy_battery_discharged') | float) + (states('sensor.energy_battery_charged') | float) }}"
                                                                                
  - platform: integration                                                       
    name: energy_battery_charged                                                
    source: sensor.power_battery_charging                                       
    method: left                                                                
    unit_prefix: k                                                              
  - platform: integration                                                       
    name:  energy_battery_discharged                                            
    source: sensor.power_battery_discharging                                    
    method: left                                                                
    unit_prefix: k   
```

ministryofsillywalks, 2021-10-04:
Thanks a lot for sharing!
However how can I add these to Home assistant energy monitoring, as it expects kWh and not W.

mpredfearn, 2021-10-04:
@ministryofsillywalks All of the "power" sensors I use to populate a https://github.com/reptilex/tesla-style-solar-power-card.
[image]
I have populated the energy dashboard as shown here:
[image]
The import / export from grid are pretty self-explanatory. The solar generation is calculated using a template from the inverter energy counter, minus the energy discharged from the battery, plus the energy charged to the battery. This gives an approximately correct "solar generation curve" taking into account the energy flowing into and out of the battery.
The battery charged / discharged sensors are generated by integrating the instantaneous power sensors (because my battery unhelpfully resets the internal charged / discharged sensors all the time). Some additional config is required to make the battery charged / discharged and solar generation sensors show up in the energy dashboard:
```
homeassistant:                                                                                                                          
  customize_glob:                                                               
    sensor.energy_solar_generation:                                                                       
      device_class: energy                                                      
      state_class: total_increasing                                             
    sensor.energy_battery_charged:                                              
      device_class: energy                                                      
    sensor.energy_battery_discharged:                                           
      device_class: energy 
 ```

ministryofsillywalks, 2021-10-04:
Again thanks for the input!
I think I found the problem. As the sun is currently shining the "battery discharged" is 0 and thus the entity doesn't show up.
Thus again the Solar Generation doesn't show up as it contains a calculation from "battery discharged".
At least thats what I think is happening, as the "battery charged" sensor does show up.
Guess I'll have to wait for evening when my battery needs to go to work.

mpredfearn, 2021-10-04:
@ministryofsillywalks ah - well I also fell into this bug: https://github.com/home-assistant/core/issues/56392#issuecomment-926429118 which resulted in the battery discharged entity not showing up in the energy dashboard, whereas the battery charged one did. I have to manually edit the database to fix it.

ministryofsillywalks, 2021-10-04:
I'm using postgres as recorder.
Funnily only the one sensor is actually showing up in statistics_meta
[image]

mpredfearn, 2021-10-06:
@ministryofsillywalks did you change it's device class to "energy" and state class to "total increasing"?

ministryofsillywalks, 2021-10-06:
Just tried that. Set it to device_class energy from the UI but the UI doesn't give me an option to change state_class to total_increasing but only to "measurement"
[image]