binsentsu/home-assistant-solaredge-modbus #45: LG Resu Prime 10H Battery show discharging at 32768W when sleeping

verstrepeng, 2021-09-22:
First of all, many thanks for this integration, it was the missing piece !
I have 1 issue, which is not directly related to this integration, as it gives the same result in the SolarEdge app itself, but I was wondering if anyone else has this issue and maybe we can cover it within the code ?
Once my battery goes into sleep mode, it shows up like it is discharging at 32768W, se rest of calculated values are wrong...

mpredfearn, 2021-09-22:
Hi @verstrepeng - I have an LG Resu 7H battery, and it shows bogus values for voltage & current when sleeping - so I added a check to the code here https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/master/custom_components/solaredge_modbus/__init__.py#L758.
Perhaps we need to set power to 0 as well.

mpredfearn, 2021-09-22:
@verstrepeng also I'd be interested in how your "battery charged" and "battery discharged" sensors behave? On my 7kWh LG battery they seem to reset at midnight, which is not how the Solaredge documentation says they should behave - they should be lifetime values. Because they reset, some fun and games are going to be required to make the sensors work correctly with Home Assistants Energy Dashboard.

verstrepeng, 2021-09-22:
> Hi @verstrepeng - I have an LG Resu 7H battery, and it shows bogus values for voltage & current when sleeping - so I added a check to the code here [...]

If in that check you can set power to 0 as well, this problem is gone, so would be great :-).

verstrepeng, 2021-09-22:
@mpredfearn my "battery charged" and "battery discharged" values keep counting since setup, so no midnight resets...
I have a Solaredge 5000H with SESTI.