binsentsu/home-assistant-solaredge-modbus #49: Exported energy not being calculated

funzie19, 2021-09-22:
I've not be able to successfully setup the energy dashboard and while investigating my numbers they do not match up.

Solar production is correctly calculated. So is imported power from the grid. However self consumption and exported back to the grid are never calculated. Setting up a graph for generated, exported, and imported reflects my issue. Exported energy appears to always be constant.

I think the problem here lies with the exported value and why it doesn't increment. Any help would be appreciated.

[image]

Here are also a couple screen shot of my configuration and dashboard.
[image]
[image]
[image]

mpredfearn, 2021-09-23:
Hi @funzie19, so looking at your M1 imported kWh over the course of the day it looks like it is reading about 45kWh (difference between 45,759 ish and 45,706 ish). That corresponds to what the Solaredge platform is saying your consumption was through the day (45.16kWh). 
Therefore I suspect that your M1 CT is around a cable to the load, not the more common (I think) property grid connection. That means the import / export from the grid cannot be measured - they have to be calculated. Import is any deficit between production and consumption. Export is any surplus. 
See https://www.solaredge.com/sites/default/files/consumption_monitoring_na.pdf for more details - you've got a consumption meter, rather than an export / import meter.

I guess you've got 2 options
 - Move the CT to the grid connection point so that it can directly measure import / export and reconfigure the inverter to reflect that
 - Add template sensors to calculate the missing values.

funzie19, 2021-09-23:
Thanks for the reply. After I opened up this issue I did more investigating. You are correct in the fact that the meter/inverter is what is returning the static value for export. I verified this by querying the modbus directly. The CTs are in the correct location. However I will be verifying the cable connections to them, because last November I had to replace the inverter. So maybe a loose connection. 

From what I understand the export and self consumption is calculated on the Solaredge monitoring side, so it would have never caught this issue. My inverter is indeed setup as a `consumption` meter. I'll configure it as an `export` meter and see if anything changes. (might have been a config I missed when swapping the inverter)

funzie19, 2021-09-23:
I did some more investigation today. I updated the meter to export/import however it kept on adding solar production with import. After I popped off the panel I realized the issue. The solar lines are tied into the main lines about 12" before it connects to the panel. The CT clamps were after this, I moved the CT clamps before the solar tie in and now it measures correctly.

closed 2021-09-23 by funzie19

mpredfearn, 2021-09-24:
@funzie19 cool, glad you got it working :+1: