binsentsu/home-assistant-solaredge-modbus #50: Data validation while null values in communication

gmacioso, 2021-09-26:
I experience that from time to time there are missing data from modbus meter for short period of time. They are all null values (m1), while values from inverter (ac) are normal. Seems to be some issue in communication between meter and inverter.

Anyway, would I'd be possible to add simple validation of incoming data?
If value is null, then not update entity?
Issue is important for new  Energy Management, as it creates sudden jumps in data, as I recalculate SE Wh into kWh entity.

On pic you can see missing data around 5:20

[image]

gmacioso, 2021-09-26:
[image]

verstrepeng, 2021-09-26:
I have the same issue.
I created utility meters in HA and use these for the energy dashboard. No problem with those.

ChristophCaina, 2022-11-02:
can be closed

closed 2022-11-02 by gmacioso