binsentsu/home-assistant-solaredge-modbus #56: Name of the sensor for the solar production of the panels (kW)

klacol, 2021-10-10:
Hi all,
great Intregration here. I am using it, because I can get the values in a very fast time (about 1 second). That enables me to steer the charging of my electrical car. Somehow, I could one thing not figure out yet. 

I am seeking the sensor, that reports this value:

[image]

Does anybody know, how this value has to be calculated from the modbus sensor values?

mpredfearn, 2021-10-11:
Hi @klacol this was discussed last week here https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/55. The integration currently provides raw access to the values we can read from the inverter. But those sensors do not correspond to what you see in the Solaredge app. For that (currently) some templated sensors are necessary to do the maths. Sure it would be awesome to provide the sensors directly from the integration, but that becomes very complex very quickly due the variation on peoples systems - whether there are meters, whether there are batteries, what configuration the meters have, etc.

I gave my templated sensors in the above issue, which depending on your systems configuration (if it matches mine) may give you access to what you want. @binsentsu maybe we could turn on the Wiki for this repository so that documentation such as that can be captured in a more permanent place?

binsentsu, 2021-10-11:
@mpredfearn Agree for wiki page. Version within https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/55#issuecomment-938472190  is still correct?

mpredfearn, 2021-10-12:
@binsentsu yeah, that's still my latest version.

klacol, 2021-10-14:
Hi @mpredfearn , thanks for the clarification. Indeed, while creating the above figure I realized, that there are many installation scenarios, that are handled by the Solaredge app in the background. And the installation scenarios rule the formulas for the calculated sensors.

binsentsu, 2021-10-18:
@mpredfearn done

klacol, 2021-10-19:
Great, and Thank you! 

These template values can be used with the Tesla card, that you have descibed; but also with the [power-distribution-card](https://github.com/JonahKr/power-distribution-card).

It it worth to mention, that we have sensors, that report three different types of values:

1. The **actual current** of the systems, measured in the unit Ampere (A)
2. The **actual power** of the systems, measured in the unit Kilowatt (kW)
3. The **overall energy** of the systems, measured in the unit KilowattHours (kWh). This is the energy of a device, that runs one hour with the power of 1 Kilowatt. (Sorry, just for documentation, because I have seen so manx mixtures of them.)

I personaly use the three values as follwoing:

1. Current: Measurement of the live load (and overload) of my electrical fuses (and sometimes to cut of unneeded laods to prevent fuses from jumping out).
2. Power: Measurement of the live power (and overpower)  that my grid connecetor allows for my building
3. Energy: Measurement of the energy, that I produce, or that I have to buy.

Current and Power relate to each other very easy, through the formula `Power = Current * Voltage`; so I could also calculate the current from the power, but I often want to see the clear values as Ampere. That is, why I calculate them seperatly.

Thanks to the wiki docs, it is now easy to calculate the three values for all scenarios.

closed 2021-11-12 by klacol