binsentsu/home-assistant-solaredge-modbus #57: Feature Request: Change StorEdge profile in automations

iNaiks, 2021-10-13:
Hi,
There is any way to configure an automation for change the setting of "Maximize self consumption" to "Charge from PV and AC" at certain hour (like every day at 21:01)?

This is because my electricity tariff lets me 2 hour of free power, and in these hours I use to charge the battery (normally from 21:00 to 23:00).
Now I'm done it manually. There is any way to do it automatically? I can't see options on automations on home assistant.

Very very thanks.

binsentsu, 2021-10-16:
@mpredfearn I'm labeling this also for documentation wiki https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/57#issuecomment-942291325

mpredfearn, 2021-10-19:
Added to new wiki page https://github.com/binsentsu/home-assistant-solaredge-modbus/wiki/Automating-change-of-Storedge-Profile

ministryofsillywalks, 2021-11-05:
Just read the wiki and of course had to start experimenting.
However my setting was set to "off" which can't be as the battery is charging devices in my home.
Switching the setting to "Maximize self consumption" doesn't seem to work, as after a few seconds the setting just reverts back to "off"
Any idea how to troubleshoot this?

iNaiks, 2021-11-15:
Yes, in the readme.md it says:
"Appendix B of the Solaredge power control document gives the necessary steps to allow changing the charge / discharge mode of the battery, but essentially all that you need to do is change the "Storage Control Mode" selector to "Remote" (it is usually set to "Maximise Self Consumption") and then select a mode using the "Storage Default Mode" selector."