binsentsu/home-assistant-solaredge-modbus #69: Modbus entity unkonw

meden199, 2021-11-06:
I have installed the HACS component.
Add the solarEdge to the intergrations. The devices are showing up, but all data is unknow.

Where dit I make a mistake? was it at the Modbus RS458-2 configuration?
Maybe a screenshot from a working device will help me?

ChristophCaina, 2021-11-08:
Hi,
I've set up the RS458-2 to "SolarEdge Device"
But I will try to provide a Screenshot from my inverter Settings later today

kazek127, 2021-11-10:
I also have issue with unknow device. I can ping my inverter. But from logs I see that component can not logon to SE inverter.
Ten błąd pochodzi z niestandardowej integracji.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:215
Integration: SolarEdge Modbus (documentation)
First occurred: 17:59:12 (610 occurrences)
Last logged: 19:44:58

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 228, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 552, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=1, address=40071, count=38)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 215, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 106, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.30.8:1502)]

ChristophCaina, 2021-11-10:
Hi,
sorry - due to late shift I was not able to get the data from the inverter to show my settings :-(

If you login to your inverter with SetApp, you should be able to change the settings for RS458-2.
Please post some screenshots about your settings in RS458-1 and RS458-2 this could help trying to solve your connection issues.

kazek127, 2021-11-10:
I check the inverter again and nothing changes on it, only perform update OS of the inverter. Earlier work fine.

ChristophCaina, 2021-11-10:
I am pretty sure, that this issue is not related with the integration itself.
Most probably it is something with your modbus settings on the inverter.

It might be, that you need to change the settings for RS458-2 temporariyl and then change them back to your original configuration.
Honestly speaking, I am not quite sure about this issue - Since you haven't provided screenshots of your settings, I cannot really compare them with my setup...

HyperActiveJ, 2021-11-10:
If it's any value, last time I upgraded I had to do what  ChristophCaina suggested to get it to open the port = change the ra485 settings then change them back again. I went and actully changed all the settings when it didn't work the first time just changing -2.

kazek127, 2021-11-10:
But from SE application I can see that the inverter is connecting to the power meter via RS485. I attaching screens from my inverter
[image]
[image]
[image]
[image]
[image]

ChristophCaina, 2021-11-10:
The First Connection rs458-1 seems to be OK, thus is the Connection to the Meter.

The important Connection for the Integration should be the rs458-2.
As far as I got from the Translation, the settings in rs458-2 should also be OK.

I would try to Change it to Solarlog and then Change it Back to the current setting.

Giolovin, 2022-02-25:
I'm experiencing the same error. Has this been resolved? I'm on the latest firmware

[podrzędny](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwj0x6vQnJv2AhUFP-wKHawwDygQFnoECA0QAQ&url=https%3A%2F%2Fpl.wiktionary.org%2Fwiki%2Fpodrz%25C4%2599dny&usg=AOvVaw08YGqVLusriSSlaeJPWof7)

Giolovin, 2022-02-25:
These are my settings (in Dutch)

I'm connected with a powerline adaptor with a lan + fixed ip address on the inverter
[image]

kazek127, 2022-02-25:
I still have this issue. After the last update, Homeassistant SE stops working with HA. SE is pinging but I got error that HA can not connect to SE

purcell-lab, 2022-03-15:
Are you connecting to the inverter over wifi?  ModbusTCP over wifi seems to be an issue with the new firmware.

theneiljohnson, 2022-05-04:
I've got this issue too, can anyone help please?
[image]
[image]

theneiljohnson, 2022-05-04:
I managed to fix my issue by disabling Modbus TCP and then re-enabling again. Very weird!

enry86cami, 2022-07-23:
Hi all, on HA I read this:
```
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 228, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 552, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=1, address=40071, count=38)
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 215, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/srv/homeassistant/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/srv/homeassistant/lib/python3.9/site-packages/pymodbus/client/sync.py", line 106, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.118:1502)]
```
This is my conf:
[image]

I'm connected by wifi...