binsentsu/home-assistant-solaredge-modbus #71: Autoamtion: Change storage_remote_charge_limit depending on m1_ac_power

iNaiks, 2021-11-15:
Hi, I'm using an automation to change the value of "storage_remote_charge_limit" depending on value of "m1_ac_power".

Now I'm doing depending on number value of "m1_ac_power" for certain time and change the value of "storage_remote_charge_limit" at certain value.
Like "m1_ac_power" is between -2500 and -3500 set "storage_remote_charge_limit" to 3000

There is any possibility to automatically change the value depending on the variations of "m1_ac_power" depending on my home usage.

Explain example:
My maximum consumption of the grid is 6kWh.
Depending on the "m1_ac_power" value, change the "storage_remote_charge_limit" to stay nearby 6kwh.
The value of "m1_ac_power" changes on time depending on my home usage. The difference from my home usage and 6kWh is the value to set in "storage_remote_charge_limit".

There is any way to do that?

binsentsu, 2021-12-14:
I think it's possible through service Number:SET and then selecting the storage_remote_charge_limit entity.
[image]
@mpredfearn Can you confirm if this assumption is correct?

iNaiks, 2022-02-17:
Ohh yes this was my first problem,

Now I'm using a template sensor with the calculation of used home energy and depending on that change the "storage_remote_charge_limit" value.

The calculation for consumption of my house is (in my case with battery LG CHEM Resu10h):

house_consumption:
  friendly_name: 'House Consumption'
  value_template: " {{ states('sensor.solaredge_ac_power') | float(0) | round(0) - states('sensor.solaredge_m1_ac_power') | float(0) | round(0) }} "
  unit_of_measurement: "W"
  device_class: power
  
  And work pretty nice.

closed 2022-02-17 by iNaiks

mpredfearn, 2022-02-17:
Cool :+1: perhaps you could add a page to the Wiki documenting what you've done and why, to help out those that follow?
Thanks :-)
Matt

iNaiks, 2022-02-17:
Yes of corse,
How can I do that?
Adding a post on wiki?
Can I add a video with instructions?
Thanks
Nico