binsentsu/home-assistant-solaredge-modbus #73: UI configurator doesn't show up

borgswen, 2021-12-10:
Hi,

This may sound stupid but I cannot find what I am doing wrong. I have installed the repository in hacs, restarted home assistant but now it doesn't show up in the menu with integrations I can set up in the integrations tab.
Any ideas as to what might be going wrong would be much appreciated.

Borg

binsentsu, 2021-12-10:
Could be browser caching. Try:
Windows: ctrl + F5
Mac/Apple: Apple + R of command + R
Linux: F5

borgswen, 2021-12-11:
Thanks a lot, that solved the issue

Ventusfahrer, 2021-12-11:
I had the same problem and die solve not by pressing F5 (Linux). instead I did invoke (GUI) Home-Assistant - Settings - Integrations -Add Integration, select solaredge_modbus  - then the configuration dialog did pop up.

But - by the way. How do I reconfigure solaredge_modbus after it was configured successfully?

binsentsu, 2021-12-12:
There is no reconfigure UI option at the moment. Just use remove and re-add.

closed 2021-12-14 by binsentsu