binsentsu/home-assistant-solaredge-modbus #74: Issues with energy integration and sensors

atze09, 2021-12-14:
Hello i use your integration for a long time and now wanted to integrate it into the energy dashboard of HA.
But i can only make the single phases to be visualized:
[image]

The imported/exported sum's and the PV_AC is not beeing displayed.
[image]

Maybe anyone can help me out.

binsentsu, 2021-12-14:
Don't think this is something the integration can control. All kWh sensors have the correct state class and device class, so they are selectable for putting on the dashboard. The fact that they are not visible on the dashboard is out of control of this integration. New sensors take 2 hours before they appear, I assume you took that into account? What happens if you put less sensors on the board (eg. remove a,b,c and only keep the sum)

atze09, 2021-12-15:
Yes i took that into account, i also tried with less esnsors, but it doesn't work in the current month.
But if i scroll back to August it works with your sensors.
[image]

But in Septembera dn Oktober it showed the same values every day:
[image]

And in November it stopped working,
[image]

Maybe it is really caused by HA.
I will check the possibility to reset the HA energy dashboard

ChristophCaina, 2022-11-02:
not sure - we haven't heard anything back since december last year... this could probably be closed?

atze09, 2022-11-02:
yes

closed 2022-11-02 by atze09