binsentsu/home-assistant-solaredge-modbus #75: Error in logs

pepsonEL, 2021-12-14:
Hi. I installed integration , enable modbus and port in my inverter and then start configuration in HA.... add IP and port and no read data. Show me entities but without data,..... In logs i have this error:

Ten błąd pochodzi z niestandardowej integracji.

```
Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:215
Integration: SolarEdge Modbus (documentation)
First occurred: 12:05:09 (4 occurrences)
Last logged: 12:06:39

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 228, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 552, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=1, address=40071, count=38)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 215, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 108, in execute
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.4.21:1502)]
```

KM192, 2021-12-14:
How do you connect to Inverter - by LAN or WiFi ?

pepsonEL, 2021-12-14:
WIFI. But some days ago support from solaredge update me firmware. Before works ok. I think that update destroy communication. Modbus and port in setting inverter is enable.

KM192, 2021-12-14:
In the newest firmware Solaredge has blocked Modbus interface via WiFi. I had the same issue with the same error. You have to downgrade (short-term) or move to cable LAN. 
And escalate that so maybe they will change this strange decision.

pepsonEL, 2021-12-14:
But how downgrade FW ?

KM192, 2021-12-14:
Only by contacting the support.

pepsonEL, 2021-12-14:
But did you know when i factory reset my inverter i can connect to it by setapp without my seller which now controller my instalation ? And after reset factory i must specific setup my inverter or only create connection wifi ?

pepsonEL, 2021-12-14:
For me better is WIFI.... becuase i dont have any router or LAN port about Inverter

pepsonEL, 2021-12-14:
But where is LAN port in inverter ?

KM192, 2021-12-14:
inside the case

pepsonEL, 2021-12-15:
Support downgrade me Firmware and integration now works ok.

scottda4270, 2022-01-18:
Thanks for this. Solaredge updated my firmware too and the modbus integration stopped working two weeks ago. Could not figure it out. Just Ethernet cables the inverter and now working fine again!!! Brilliant. Thanks

Sven-v-Beuningen, 2022-01-22:
That's interesting. I've got a solaredge inverted installed a few days ago and since then I try to get the solaredge modbus connecttion to work. I don't see any errors, but all sensors in HA show "unknown" state. 
During installation of the inverter, I also did get a firmware update.

scottda4270, 2022-01-24:
@Sven-v-Beuningen ...do you have Modbus TCP enabled on the inverter...mine was disabled by default and i had to enable it following the instructions (i.e. by connecting to inverter using app, connecting a browser to the inverter then changing settings from there...its a well documented routine). thanks

Sven-v-Beuningen, 2022-01-24:
@scottda4270 Yes, I also had to activate modus TCP. I think I can see some kind of communication in the logs, but I don‘t know if the data is right. There are no errors in the logs.

Best regards

Sven

scottda4270, 2022-01-26:
Sorry @Sven-v-Beuningen , not sure i can offer you any help on this. My knowledge is pretty minimal on this so not able to offer any extended debugging assistance. Hope you manage to get it sorted soon. thanks

Sven-v-Beuningen, 2022-01-26:
> Sorry @Sven-v-Beuningen , not sure i can offer you any help on this. My knowledge is pretty minimal on this so not able to offer any extended debugging assistance. Hope you manage to get it sorted soon. thanks

I just found the solution to the issue. I configured the integration to use multiple meters, but I just have one. I added some logs to the code, so that I could see where the issue occurs. After removing meter 2 and 3, everything works fine. 
Maybe some logging output would be helpful, because in my logs every thing was ok.

scottda4270, 2022-01-26:
Ah ok. Glad you got it working in the end. It’s pretty good and has enabled me to create a dashboard in Grafana with everything I need. Thanks

ChristophCaina, 2022-11-02:
@binsentsu - this could probably be closed :)

closed 2022-11-09 by binsentsu