binsentsu/home-assistant-solaredge-modbus #78: ValueError: Value 0 failed validation (>0)

Cereal2nd, 2022-02-03:
V1.5.1
meter 1 and battery 1 enabled

```
2022-02-03 15:08:49 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 229, in read_modbus_data
    and self.read_modbus_data_meter1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 241, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 405, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)
```

mpredfearn, 2022-02-03:
The issue here is that meter1 is reporting that 0 kWh has been exported by the system to the grid in it's lifetime, which is extremely unlikely.

Exactly the same issue was reported in https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/17 and in the end it looked like something might be wrong / unexpected in the Inverter configuration because https://github.com/nmakel/solaredge_modbus/ - which uses the same modbus registers as this integration - was showing no valid meters, even though the Inverter UI was.

Please could you try https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/17#issuecomment-1010919930 and report back?

I'm not sure if @pczolee was able to resolve the issue via SolarEdge support.

ChristophCaina, 2022-11-02:
should be fixed with #114

closed 2022-11-30 by binsentsu