binsentsu/home-assistant-solaredge-modbus #79: After update HA to 2022.02 connection error

kazek127, 2022-02-13:
Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 228, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 552, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=1, address=40071, count=38)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 215, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 106, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.30.8:1502)]

From the machine where is installed HA I can get access to 192.168.30.8

Pablo1, 2022-02-13:
Same here. I'm using port 502 but the error log shows 1502.

purcell-lab, 2022-03-15:
Are you connecting to your SolarEdge inverter over the Wi-Fi interface?

This appears to be no longer supported with new firmware versions.

kazek127, 2022-03-15:
yes, that is correct. I call to SE support and they after update inverter block that option via wifi :/

pepsonEL, 2022-03-15:
Yes in new FW modbus over WIFI was blocked.

stuarts0011, 2022-05-10:
I asked solaredge to downgrade the firmware if SE4000 inverter to 4.12.35
and modbus over wifi now works !