binsentsu/home-assistant-solaredge-modbus #80: Error reading modbus data

anbraun123, 2022-03-24:
Hi,

my recently installed Solaredge SE10K is connected via LAN (no WiFi). 
I´m able poll Modbus Data via mbpoll commandline on port 1502:

Protocol configuration: MODBUS/TCP, FC3
Slave configuration...: address = 1, start reference = 40070, count = 1
Communication.........: x.x.x.x, port 1502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[40070]: 103
-- Polling slave... (Ctrl-C to stop)
[40070]: 103
-- Polling slave... (Ctrl-C to stop)

In HomeAssistant I receive the status "unknown", the Log shows me this message:

2022-03-24 10:54:59 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
update_result = self.read_modbus_data()
File "/config/custom_components/solaredge_modbus/__init__.py", line 228, in read_modbus_data
self.read_modbus_data_inverter()
File "/config/custom_components/solaredge_modbus/__init__.py", line 627, in read_modbus_data_inverter
acenergy = validate(self.calculate_value(acenergy, acenergysf), ">", 0)
File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

Any guidance or tipps? Would love to use to visualize my system, the bloody SolarEdge App is a mess when we talk about data...

sragas, 2022-04-05:
Same for me. 
Fork [solaredge-modbus-multi](https://github.com/WillCodeForCats/solaredge-modbus-multi) is running fine. But no Battery support. Would like to use this one.

randomstring, 2022-06-13:
I'm getting a similar error:
`Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 229, in read_modbus_data
    and self.read_modbus_data_meter1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 241, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 405, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")`

Two inverters, one leader and one follower. The leader has the ethernet connection. I also have a battery.

It's not clear to me what options to select when setting up the the plugin. Is there something that explains the setup for different solaredge configurations?

tias2001, 2022-06-25:
My installarion started this error overnight, stopped trying to poll the inverter at all.. not sure what caused it, I tried reloading the integration and restoring partial backups to no avail.  A full backup restore from the night before however got it back online..

ChristophCaina, 2024-01-06:
initial error seems to be the same as #191 and the others