binsentsu/home-assistant-solaredge-modbus #82: Multiple connections to modbus possible?

ministryofsillywalks, 2022-04-17:
Hi,
the integration is running really well in HA, however I want to integrate solar charging an ev and would like to use evcc.io for this.
However when trying to connect evcc to the solaredge modbus I get the error that the connection was refused.
I'm guessing this is because solaredge only accepts one connection via modbus and not multiple?
Just wanted to check if someone can confirm this? If so then I will have to make a choice between this integration and evcc... :(

sebnaf, 2022-04-17:
Correct. 1 concurrent connection only. 
Use evcc it let it push data to mqtt - et voila.

ministryofsillywalks, 2022-04-17:
Thanks!
I guessed this would be the way to go. Will have to redo quite a lot when changing the sensors but it is what it is ;)

sebnaf, 2022-04-17:
At least evcc -> mqtt runs really stable.

ministryofsillywalks, 2022-04-18:
> At least evcc -> mqtt runs really stable.

do you have it setup like this?
Could you share your config of getting the data from mqtt into ha?

sebnaf, 2022-04-19:
Sure, hope this helps

```
uri: 0.0.0.0:7070 # uri for ui
interval: 15s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: <redacted>

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: error

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters

meters:

- name: grid
  type: movingaverage
  decay: 0.1
  meter:
    type: custom
    power:
      source: modbus
      model: sunspec
      uri: 10.10.20.7:502
      id: 1
      subdevice: 1
      value: 203:W
      scale: -1

- name: pv
  type: custom
  power:
    source: modbus
    model: sunspec
    uri: 10.10.20.7:502
    id: 1
    value: 101:W

chargers:
- name: elektronenpumpe
  type: go-e
  uri: http://10.10.20.22

vehicles:

- name: tesla
  type: tesla
  title: <redacted>
  capacity: 82
  tokens:
    access: <redacted>
    refresh: <redacted>
  vin: <redacted>
  cache: 10m

# site describes the EVU connection, PV and home battery
site:
  title: ohw3 # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Elektronenpumpe # display name for UI
  charger: elektronenpumpe # charger
  #meters:
  #  charge: charge # charge meter
  vehicle: tesla
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  # - ID.3
  # - e-Up
  mode: pv
  resetOnDisconnect: true 
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 120m
    min: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 80 # always charge to 100%
    estimate: false # set true to interpolate between api updates
  #onDisconnect: # set defaults when vehicle disconnects
  #  mode: pv # switch back to pv mode
  #  targetSoC: 80 # charge to 100%
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: -1500 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 100 # maximum import power (W)
  guardduration: 15m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.24 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.065 # [currency]/kWh

mqtt:
  broker: 10.0.55.22:1883
  topic: #evcc-push
  user: mqtt
  password: mqtt

influx:
  url: http://10.0.55.22:8086
  database: evcc
  user: evcc
  password: evcc

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  - type: pushover
    app: <redacted>
    recipients:
    - <redacted>

```

ministryofsillywalks, 2022-04-19:
Thanks for that. However I meant the intergation of evcc on the HA side of things.
I figured this out in the meantime though:
```
# evcc template
  - sensor:
      - name: "EVCC Battery Power Charging"
        unit_of_measurement: W
        device_class: power
        state: "{{ max(0, 0 - states('sensor.evcc_battery_power_w') | float(default=0)) }}"
      - name: "EVCC Battery Power Discharging"
        unit_of_measurement: W
        device_class: power
        state: "{{ max(0, states('sensor.evcc_battery_power_w') | float(default=0)) }}"
      - name: "EVCC Energy Export"
        unit_of_measurement: W
        device_class: power
        state: "{{ max(0, 0 - states('sensor.evcc_grid_power_w') | float(default=0)) }}"
      - name: "EVCC Energy Import"
        unit_of_measurement: W
        device_class: power
        state: "{{ max(0, states('sensor.evcc_grid_power_w') | float(default=0)) }}"
      - name: "EVCC PV Generation"
        unit_of_measurement: W
        device_class: power
        state: "{{ max(0, states('sensor.evcc_pv_power_w') | float(default=0)) }}"
```
```
#EVCC integration
  - platform: integration
    source: sensor.evcc_battery_power_charging
    name: "EVCC Total Battery Energy Charged"
    unit_prefix: k
    method: left
  - platform: integration
    source: sensor.evcc_battery_power_discharging
    name: "EVCC Total Battery Energy Discharged"
    unit_prefix: k
    method: left
  - platform: integration
    source: sensor.evcc_energy_import
    name: "EVCC Grid Import"
    unit_prefix: k
    method: left
  - platform: integration
    source: sensor.evcc_energy_export
    name: "EVCC Grid Export"
    unit_prefix: k
    method: left
  - platform: integration
    source: sensor.evcc_pv_generation
    name: "EVCC PV Generation kwh"
    unit_prefix: k
    method: left
```

ChristophCaina, 2022-06-02:
Hi, there's an option on how to establish multiple connections via Modbus :)
I recently found this, because my wallbox also required a modbus connection to the inverter.

There's an addon available for HomeAssistant:
https://github.com/Akulatraxas/ha-modbusproxy

when you setup this addon, you can connect this to your inverter - and then, configure solaredge_modbus and other devices to the IP of your HomeAssistant instance.

The addon will then forward the communication from the Modbus Device to each client connected.

vanderdude, 2022-06-25:
> Hi, there's an option on how to establish multiple connections via Modbus :) I recently found this, because my wallbox also required a modbus connection to the inverter.
>
> There's an addon available [...]

Would this also be the way to go if you have 2 inverters?

ChristophCaina, 2022-06-25:
I don't know, if this could work if you are running two inverters.
I think, that strongly depends on how these inverters are connected.

https://www.solaredge.com/sites/default/files/solaredge-communication_options_application_note_v2_250_and_above.pdf

vanderdude, 2022-06-26:
They are being installed in 2 weeks and I was planning on just connecting both to my wired Ethernet. Then enable Morbus over TCP and run this addon on my HA. But is there any provision to collect data from 2 IP-adresses?

Noolix, 2022-09-25:
I have the same problem, that I have other clients, that want to access the inverter via Modbus TCP as well. They both access the inverter by connecting, passing the request, getting the result, disconnecting. They work togther. if it happens, that the connection is blocked by one and the other wants to access, it gets an error. So it waits some time and retrys. Eventually, it will get its result.

It would be great, if home-assistant-solaredge-modbus could do this as well. The implementation in the current state is not usable in a more complex environment with several devices accessing the inverter.

ChristophCaina, 2022-09-25:
Hi @Noolix 
You can try to use the HomeAssistant Addon "Ha Modbus-Proxy" (https://github.com/Akulatraxas/ha-modbusproxy)

With this Addon, you can connect to your SolarEdge Inverter - and then, all other applications and clients can connect to the Modbus-Proxy.
This is working perfectly fine for me - while this integration - as well as my Wallbox and an additional Sunspec integration are connected to the proxy, instead of the inverter itself.