binsentsu/home-assistant-solaredge-modbus #84: Battery entities showing abnormal values

treynaer, 2022-04-28:
Hi,

Following battery entities sporadically show abnormal values:
sensor.solaredge_battery1_charged
sensor.solaredge_battery1_discharged
sensor.solaredge_battery1_current
sensor.solaredge_battery1_voltage

Can you add some logic to filter this out? I think sometimes the connection with the battery is lost for the read cycle but is reachable again the one after.

Regards,
Tim

[image]
[image]
[image]
[image]

mkrasselt1, 2022-10-11:
its the same with outer values - if the inverter dosnt answer in time or the values are somhow corupted,
the calculations default to zero. 
there is no one click solution though as we need to detect wrong/unlikely values and deliver unavailable/last value instead

ChristophCaina, 2022-11-02:
this might have been fixed with pr #112 / #114