binsentsu/home-assistant-solaredge-modbus #85: Control of export limit and production limits

ThirstyDursty, 2022-05-10:
Hi

I see this component can control the battery settings

Is it possible for it to also control the export limit (set to zero export for when feed in tariffs go negative)
and production limits to zero for when pricing goes negative...

I'm with Amber energy and regularly we get negative feed-in/export pricing and occasionally we get negative energy pricing (today had 2 hours of negative pricing (paid to consume) and 30min of that was 0.05/kw)

djfanatix, 2022-06-01:
Yeah would also like this feature.
You can control this with parameter F001
[https://tweakers.net/i/_Vbw6h07uP3EtmVy-RP_ZXr0jnM=/full-fit-in/4000x4000/filters:no_upscale():fill(white):strip_exif()/f/image/IHKtvheNjManYE9TaabWoDh8.png?f=user_large](url)

iNaiks, 2022-07-24:
Hi, I'm interested to change the export limit. How can I do with parameter F001?

tias2001, 2022-07-24:
This pull request allows it, works very well
https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/42

iNaiks, 2022-07-24:
Hi thanks, bit i don’t understand how to set or configure. 🥲

tias2001, 2022-07-24:
You just need to overwrite the files from that pull request into config/custom_components/solaredge_modbus
Then restart home assistant
You will then have a few more entities available and to set export limiting on you need to set both export control mode to Direct export limitation and export control site limit to 0
I.e
service: select.select_option
data:
  option: Direct Export Limitation
target:
  entity_id: select.solaredge_export_control_mode

service: number.set_value
data:
  value: '0'
target:
  entity_id: number.solaredge_export_control_site_limit

To set it back you just need to reset the control mode
service: select.select_option
data:
  option: Disabled
target:
  entity_id: select.solaredge_export_control_mode

The number will reset itself

iNaiks, 2022-07-25:
Hi, I've downloaded a folder named "home-assistant-solaredge-modbus-mpredfearn-add_export_control".
Inside on /custom_components/solaredge_modbus, there are the files that a need to copy. (Right?)

My question is, I lose any of configurations of my entities? Like name (changed manually) or disabled entities?

Very very thanks!

Nico

tias2001, 2022-07-25:
You shouldn't no but backup your files and backup home assistant

ThirstyDursty, 2022-08-02:
> You just need to overwrite the files from that pull request into config/custom_components/solaredge_modbus Then restart home assistant You will then have a few more entities available and to set [...]

Do these entities show up on their own? I still only have 83 entities and none are select. All are sensors

tias2001, 2022-08-02:
binsentsu just updated the master with this so redownload from hacs and
they should be there now

tias2001, 2022-08-02:
oops sorry looks like it hasnt been released yet so wait until the version
gets released

binsentsu, 2022-08-02:
1.6.0 has just been released

Ralphy140, 2022-08-03:
Just upgraded to v1.6.0 and re-added the integration and don't see any new controls or options to limit export in the integration. Am i missing something or did it not make the code for the 1.6.0 release ?

ThirstyDursty, 2022-08-03:
> Just upgraded to v1.6.0 and re-added the integration and don't see any new controls or options to limit export in the integration. Am i missing something or did it not make the code for the 1.6.0 release ?

Update worked for me. Couple new select.solaredge….entities

goingbto test today

Ralphy140, 2022-08-03:
would you mind popping in a screenshot just to make sure i am not an idiot :) just to make sure i know where i am looking

AlexandreLogist, 2022-08-05:
Hi all,

Deleted my integration and added it again, but no extra select entities.
My converted supports the throttling of export, so I guess it should be possible to use in HA.

Any other limitations that could explain why I don't have the controls? 

Kind regards,
Alexandre

Ralphy140, 2022-08-05:
> Hi all,
> 
> V1.6.0
> 
> Deleted my integration and added it again, but no extra select entities. My converted supports the throttling of export, so I guess it should be possible to use in HA.
> 
> Any other limitations that could explain why I don't have the controls?
> 
> Kind regards, Alexandre

Hey Alexandre, 

I suspect you are having the same "issue" I had. The new entities are not under the integration its self. you will need to go to developer tools and under the states if you search you should see a new entity called "solaredge (Export control mode)" 

I have written a small automation to turn on and off the export controls based on a binary helper that you can toggle on a dashboard if you want. Or if you want to get smart and you have real time feed in tariffs with something like Amber in Australia you could set the export limit to 0 when the value of exported power is negative 

Below are the two automations i have written 

Turn on export limit and set it to 0 export 

````````````````````````````````````````````
alias: Solar - Set Export to 0
description: ''
trigger:
  - platform: state
    entity_id:
      - input_boolean.solar_export_limit
    to: 'on'
    for:
      hours: 0
      minutes: 0
      seconds: 0
condition: []
action:
  - service: select.select_option
    data:
      option: Direct Export Limitation
    target:
      entity_id: select.solaredge_export_control_mode
  - service: number.set_value
    data:
      value: '0'
    target:
      entity_id: number.solaredge_export_control_site_limit
mode: single

Turn off export limit when solar_export_limit is turned off 

````````````
alias: 'Solar -  Export no limit '
description: ''
trigger:
  - platform: state
    entity_id:
      - input_boolean.solar_export_limit
    to: 'off'
condition: []
action:
  - service: select.select_option
    data:
      option: Disabled
    target:
      entity_id: select.solaredge_export_control_mode
mode: single 
`````````````````````````

iNaiks, 2022-08-21:
> Fantastic, it worksss!!!. But... 😆 In my case, the voltage of the grid is very high, I only can reduce using a little amount of energy of the grid. I tried to add negative value to export control (in settapp works) for recive always 100~300w from the grid. But I can not set negative numbers. Any idea?
> 
> Super Thanks Nico

Hi, any idea for this?

ThirstyDursty, 2022-09-18:
> Hey Alexandre,
>
> I suspect you are having the same "issue" I had. The new entities are not under the integration its self. you will need to go to developer tools and under the states if you search [...]

everytime I set the export limit...it restores it automatically to 15000

tias2001, 2022-09-18:
Hi,

Have you enabled it on the inverter?

https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/42#issuecomment-924407891

purcell-lab, 2022-09-18:
> everytime I set the export limit...it restores it automatically to 15000

I'm having the same problem. When I set a new value it works for about 30 seconds and then resets back to 50000.

I have confirmed with Amber they are no longer sending commands to SolarEdge inverters until the reset big is fixed.

purcell-lab, 2022-09-18:
When I try to disable export limitation with this integration it automatically switches back.

Do I need to change something in the inverter setup before I can use this integration to change modes?

[image]

tias2001, 2022-09-18:
Hmm interesting.. mine has set export limit for a good portion of the day
today.. hopefully this isn't a bug that comes my way.. what inverter
firmware do you have installed?

purcell-lab, 2022-09-18:
> Hmm interesting.. mine has set export limit for a good portion of the day today.. hopefully this isn't a bug that comes my way.. what inverter firmware do you have installed?

I don't think this is a bug, I'm just trying to workout howto do setup, I haven't had this working yet.

I have some funny firmware revisions and with the reintroduction of modbusTCP over wifi I'm wondering if a firmware upgrade maybe worth the risk.

```
- name: Inverter 1
manufacturer: SolarEdge
model: SE7K-AUBTEBEU4
communicationMethod: RS485
cpuVersion: 4.14.228
SN: XXXXXX-XX
connectedOptimizers: 25
- name: Inverter 2
manufacturer: SolarEdge
model: SE7K-AUBTEBEU4
communicationMethod: ETHERNET
cpuVersion: 4.15.110
SN: XXXXX-XX
connectedOptimizers: 25

purcell-lab, 2022-10-02:
I have checked with SolarEdge why my export controls are constantly resetting and they advised that you cannot change the Export limit configuration while your system is subscribed to grid services.

I have enrolled my inverters with Amber Electric SmartShift as they offer remote curtailment (zero export) during periods of negative FIT. They have had to suspend this as they were seeing the inverters reboot when changing settings and have stopped issuing remote commands, but it sounds like I might need to un-enroll. 

I am now seeking to get my inverters un-enrolled so I can set zero export locally.

I'm not sure if there is a ModBus register that can check for grid services?

@ThirstyDursty you may want to confirm with your setup.

ThirstyDursty, 2022-10-02:
I logged onto the solaredge local set up wifi and turned off “grid services”. It didn’t seem to make any differenceSent from my iPhone

ThirstyDursty, 2022-10-11:
🧐 not sure I understand how to do that. Or is that reflective of menu items in the set-up screens of the inverter. Which I accessed to set up the tcp modbus connection. Sent from my iPhone

https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/42#issuecomment-924407891

ThirstyDursty, 2022-10-11:
> Oh right, did you follow the setup from the post I linked above?
> […](#)

I don't understand how to do that set up...How do you change those register items? I can log into the setup using AP mode on the inverter...I get options to turn on and off grid services...or turn on reactive power control etc...none of them seemed to work...

is it possible to get stepwise instructions here?

purcell-lab, 2022-10-11:
> Oh right, did you follow the setup from the post I linked above?

Does the binsentsu/integration change these settings (AdvancedPwrControlEn, ReactivePwrConfig & Commit Power Control Settings) on the inverter, or do I need to change these registers manually before the integration will setup zero export?

[image]

ThirstyDursty, 2022-10-16:
> Oh right, did you follow the setup from the post I linked above?
> […](#)

can we have some clarity on how to conduct the set up?

purcell-lab, 2022-10-17:
> can we have some clarity on how to conduct the set up?

I'm also following another thread which is trying to achieve the same outcome but with a lower level library.
nmakel:solaredge_modbus

However, without a proxy you can only run one ModBus client at a time.

https://github.com/nmakel/solaredge_modbus/issues/36

I'm still awaiting my inverters to be un-enroll from grid services so I can't test yet.

purcell-lab, 2022-10-25:
I logged a job with solaredge support via the app to get my firmware updated on my inverter to 4.16.23 to get ModBusTCP over wifi enabled, which they were able to complete remotely.

https://www.solaredge.com/aus/setapp-inverters-firmware

With this new firmware I am also able to use the binsentsu integration to control my export limit and my inverter no longer resets when I change export control modes. @ThirstyDursty

ejbatts, 2024-03-26:
This is an old thread, and i cant seem to find anything newer for SolarEdge curtailment in HA. I thought I had found a thread a few weeks ago but it now escapes me. I am hoping someone can help me set up my HA, or point me to another thread.

Also, my setup includes 3 inverters so I am wondering if the instructions above would work if only applied to the 'leader'.

 I had applied to onboard my inverters with Amber, but after five weeks of waiting I have reenabled Modbus TCP and supposedly that cant happen now. 

Mark @purcell-lab perhaps?

Ralphy140, 2024-03-27:
I cant answer on the 3 inverters and "leader" question but in terms of how to set curtailment you will need to make a new automation. I have written a blueprint for it here https://github.com/Ralphy140/Amber it may or may not help with your specific situation

ejbatts, 2024-03-27:
@Ralphy140 Thank you. I may need to experiment re the three inverters case as I believe the SolarEdge Modbus addon is for single inverters only? I thought that Mark @purcell-lab had multiple inverters which is why I was hoping he may offer the solution.

purcell-lab, 2024-03-27:
I'm runing the alternative solaredge-modbus-multi integration, which does handle the multi inverter solution well.
https://github.com/WillCodeForCats/solaredge-modbus-multi

I make the changes to the leader and my follower inverter changes the settings automatically.

I then use an automation when the feedin price goes negative I switch to zero export mode on the inverter.
[image]

ejbatts, 2024-03-27:
Thanks Mark @purcell-lab , really appreciated. Already using modbus-multi. I will need to investigate how to expose the  I1 Limit Control Mode options...I presume through SetApp? Or in one of the Integrations? Haven't looked yet...maybe tomorrow.

purcell-lab, 2024-03-27:
> Thanks Mark @purcell-lab , really appreciated. Already using modbus-multi. I will need to investigate how to expose the I1 Limit Control Mode options...I presume through SetApp? Or in one of the Integrations? Haven't looked yet...maybe tomorrow.

If you already have access via modbusTCP, ( which requires SetApp) then you should be able to enable the power control options directly from the integration (without SetApp)
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Power-Control-Options-%E2%80%90-Configuration

ejbatts, 2024-03-31:
Hmm. When I enable export control (export/import meter), and site limit zero, production curtails to zero and is not matching consumption?? When I disable export control, production slowly ramps up to normal, taking maybe a couple of minutes. Wondering if I should be using 'per phase' control?

ejbatts, 2024-04-02:
Changing **Site Limit** to 1 (W) seems to have resolved my problem above.

After disabling **Limit Control**, there is still a delay whilst production ramps back to normal (inverters resetting?).