binsentsu/home-assistant-solaredge-modbus #87: All sensors show as 'Unknown' when second battery configured

theneiljohnson, 2022-05-14:
If I configure the integration with the first 3 meters and battery 1, I get sensor data (but nothing from the battery)

[image]
[image]

I have 2x solar edge energy banks.

However, if I configure both batteries, i get unknown everywhere...

[image]
[image]

Can anyone help? I can't seem to get any useful data out of my batteries at all...

theneiljohnson, 2022-05-14:
This is what i'm seeing in the logs

  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:26 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:31 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:36 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:41 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:46 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:51 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
2022-05-14 17:23:56 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 234, in read_modbus_data
    and self.read_modbus_data_battery2()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 736, in read_modbus_data_battery2
    return self.read_modbus_data_battery("battery2_", 0xE200)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 815, in read_modbus_data_battery
  File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate

ChristophCaina, 2022-05-25:
Hi, how does your second battery communicate with the inverter?
SolarEdge does only allow one active modbus connection.
So in theory, I would expect that the second battery might use the channel which you have previously used with your homeassistant integration.

I recently had the same issue after a wallbox has been installed.
The wallbox needs to communicate with the Inverter to get proper data - since then, the communcation with the modbus integration does no longer work.

I found a solution for this, which I still need to setup and try... but if it's the same case for you, it might work:

[https://github.com/Akulatraxas/ha-modbusproxy](https://github.com/Akulatraxas/ha-modbusproxy)

You need to install this as an Addon in HomeAssistant.
Then, it will connect to the Inverter - and you could setup your battery and the modbus integration to get their data from the HA IP instead of the inverter directly.
This addon acts as a Modbus-Bridge...

wardienet, 2022-07-08:
Hi I’ve got a similar setup with SolarEdge’s CT clamp on the house output which is wired in to the inverter using RS-485 data cable. I’ve also got two Energy Bank batteries that talk to the inverter over WiFi somehow. I switched TCP on but didn’t yet connect to it. I was worried that the remote TCP connection may mess up the existing comms - I assume over modbus? - with the CT clamp or the batteries (IDs?). I thought I read in the SE docs that the modbus protocol won’t support both TCP and RS-485 connections simultaneously, but I have have misinterpreted that. Or the IDs may clash. Am I worrying unnecessarily? Anything I can check or pre-configure to try it? PS I’m not an installer so don’t have SetApp.

ChristophCaina, 2022-07-08:
Since I don't use any WiFi Communication Setup, I can't really give any advice here.
But usually, the TCP Interface of the Inverter (doesn't matter if it is wired or wifi) emulates an RS485 interface (RS485 over TCP)

It could be that connecting to the Modbus Interface over TCP, you will block the communication between your Inverter and the Batteries... but without further check on the configuration of your inverter (SetApp required) this is hard to verify.

You don't need to be an installer to use SetApp... it is possible to use it as consumer... but there is the risk of changing important system settings... and to mess the complete installation.
So Maybe, I would try to get in contact with your installer and describe your requirements to him.
Maybe, he can assist or setup the system accordingly.

If you need multiple clients connected to the same RS485 (TCP) - you can probably use the modbus proxy which I have mentioned above.
In this setup, the proxy connects to your inverter - and the batteries should connect to the proxy to get the required information from the inverter.

Then, you could connect your SolarEdge Modbus integration to the proxy, too - and it will also receive the required information.

ChristophCaina, 2022-07-13:
Probably, you don't need to change something on your Inverter.... But I don't have Batteries and I don't know your setup - so I can't really give any advice here.

What I know so far:
My Inverter just accept incomming connections, so I had to setup the proxy to connect to the Inverter and then anything else connecting to the Proxy...

Anything else is above what I can support remotely :-(

ChristophCaina, 2022-11-01:
Hi, comming back to that topic:
Do you really have three Meters installed?

Please note, that even with only one meter connected, you are getting at least an "export meter" an "import meter" a "consumption meter" and a "selfconsumption meter" implemented into the system.

So, please only select all three meters, if you really have three different hardware meters connected to the inverter.
I've noticed this in my setup - that when I activate more than one meter, the values become "unknown".

This is probably caused, because with more than one meters, the addresses from the Modbus communication might change.

theneiljohnson, 2022-12-15:
Unfortunately, i cannot seem to get this to function reliably, no matter what I try. If I select both batteries and one meter, it works, but after an hour or two, either one or both of my batteries fail to connect to the inverter and stop working. I then have to remove the modbus integration and restart the inverter and batteries to get them to function again.

I can't find any way to configure the batteries or inverter to use a modbus proxy.

I accept that this is likely a limitation within SE modbus of port restrictions, but it might be worth mentioning somewhere that if you have more than one battery, this is likely to cause issues.

theneiljohnson, 2022-12-28:
Ok, so with a bit of time over the holidays i managed to get this working semi-reliably. I had to drop the polling interval down to 300s but it's been running like that for a week or so now and is puling back data from both batteries and all of the items that I need from M1.

closed 2022-12-28 by theneiljohnson