binsentsu/home-assistant-solaredge-modbus #89: is it possible to get more information from the inverter?

ChristophCaina, 2022-05-29:
Hi,
I was going throught a couple of monitoring solutions for SolarEdge Inverters here on Github (just reading their descriptions and what they can provide)...
A few of them seems to be able to get more information back from the inverter than the specs tell us in the first place.
For example - the inverter might send information about the optimizers (if installed) - and also information about the firmware versions and serial numbers.

Basically, all the information, that can be checked within the online monitoring (and which I am polling via Node-Red from the online API at the moment)

I wonder, if it would be possible, to get these informations somehow directly from the inverter per modbus interface, too?

HyperActiveJ, 2022-05-31:
Want to share some links? Most other solutions utilize one of either: 1. The inveters local web api, which is being disabled in recent firmware updates. 2. Placing the solution as man in the middle between he inverter and the solargedge website, which is cumbersome for most users to setup since it requires more advanced networking experience and key gatherjng. Or 3. Utilizing the hard wired RS485 connection to emulate a master in a chain and grab dara that way, which requires purchas and hardware setup of rs485 to eth/wifi/usb adapter. 

If you know of something else let me know, for example the 485 and top might be more overlaping in functionality then the spec and documentation would lead one to belive.

Let me know if you  have seen something I havent.

ChristophCaina, 2022-06-02:
Hi @HyperActiveJ - right now, I don't have an direct example.
I figured out, that I can request several information directly from the Web API with a Node_Red integration, so I asume, the inverter can send several other values.

As far as I got, the inverter has two methods:
1. Modbus -> Sunspec Protocol
2. Modbus -> SolarEdge (Master)

And as I understood, the Inverter itself can provide more information which are not covered by the sunspec protocol... (and since I could use the Sunspec integration to get the same information as per the solaredge_modbus, I would like to have the added values when I use the solaredge_modbus integration running.

I will check, and try to get examples by other modbus monitoring solutions for SolarEdge - but setting this up, might take some time since I have a couple of other things to do atm...

HyperActiveJ, 2022-06-02:
I 100% hope your right @ChristophCaina), however I'm afraid that I think that your confusing "Modbus" with "RS485"/"TCP" in your method list. "Modbus" is a protocol/standards for setting and retrieving arbitrary data by address over any type of link, in our inverters case that comes down to either RS485 (serial/RTU) or TCP (networked/ethernet/wifi/ip). Over "Modbus" you can use the Sunspec 'Protocol' which is in effect is just documentation of memory addresses that can be read mapping to their consents. You can also use the "Power Control Open Protocol", which defines additional R/W addresses that you can use to control the inverter (and batteries), and since these additional addresses do not conflict with Sunspec both can be used over the same link. Neither appears to support optimizer level detail - though its possible that there are publicly undocumented addresses which do provide this information. The "Solaredge" 'proprietary' protocol  does not use the "Modbus" standards. Instead it appears to use its own custom protocol with entirely different headers and structure. Because of that, it seems like they have not enabled us to use that protocol over 'modbus TCP' port that can be opened over the network as used by this integration. Instead its only over the RS485 link then the physical RS485 port is configured to do so, or by intercepting the similar but not excatly the same TCP link used to communicate to the SolarEdge cloud server. There doesn't seem to be the possibility of having both both on the same serial/tcp port since the structures are incompatible.

Again I'm really hoping you either discover undocumented modbus address, or a way to utilize the 'solaredge' protocol over TCP without requiring use of a physical RS485 to TCP converter (which are inexpensive and easy to install BTW)!

You are missing option 3: Scrape LOCAL "Web API"  presented by the inverter! This is similar to how you grab that info from the web.  However in the US/NA at least this is now only available if you connect to the local wifi AP presented by the unit, at least with current FW revisions. There are people who have setup a Wifi "Client" on their network (such as a repurposed router flashed with OpenWRT or similar), to make that API accessible on their LAN. However I found that for me at least, sometimes the inverter does not open its wifi port for connections without physically moving the connect enable toggle on the bottom of the unit making this solution unreliable.

ChristophCaina, 2022-11-01:
Hi,

I am comming back to this topic - since I am going through the sunspec documentation from SolarEdge again...

I think, there are some values, that could be implemented into the integration:

[image]

Here, it might be the
C_Manufacturer = SolarEdge
C_Model = SolarEdge specific Value (Inverter Model)
C_Version = SolarEdge specific Value (What version? CPU Version?)
C_SerialNumber = SolarEdge specific Value => Inverters Serial No.

Also, there are some meter specific information that could be collected:
[image]

https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf

ChristophCaina, 2022-11-02:
In addition to my reply above:
These could be some information that would also help to define the status_vendor_codes more precisely:
[image]

If it would be possible to read if the device is singlephase, splitphase or threephase, we could define status_vendor_codes for the different devices... (?) (see pr #120)