binsentsu/home-assistant-solaredge-modbus #90: M1 AC Power seems to be the current generation kWh, but I can't use it as such

nikkilocke, 2022-06-03:
I want to use modbus variables for my Energy dashboard, because the sampling rate is so much higher than the SolarEdge web integration.

But I can't add the relevant variables to the Energy dashboard - they do not appear in the list.

For example, M1 AC power appears to be the current power generated by the solar panels. But it is not in the solar generation list in the Energy panel listing.

const.py has
```
    "M1_AC_Power": ["M1 AC Power", "m1_acpower", "W", None],
```

I tried changing that from None to "mdi:solar-power", and restarting HA, but that didn't help. Not sure what I'm doing, so asking your advice.

stuarts0011, 2022-06-04:
M1AC power is in kW not kWh. It's the wrong unit.

nikkilocke, 2022-06-05:
Thanks. I was hoping for a display that showed the power situation right now (in kW), rather than the energy today. Haven't got to grips with HA yet, obviously.

closed 2022-06-05 by nikkilocke