binsentsu/home-assistant-solaredge-modbus #91: Connection refused

Santanachia, 2022-06-20:
I have enabled Modbus TCP
My inverter responds for PING
Unfortunately i have error: `Connection to (192.168.8.116, 1502) failed: [Errno 111] Connection refused`

CPU version: 4.14.228, Connection type: WiFi

stuarts0011, 2022-06-20:
You need to downgrade the firmware if you want to use modbus over wi-fi

V.4.12.35. works.

Get in touch with solaredge.

Santanachia, 2022-06-22:
thank you for your suggestion. For now, they wrote me that they don't have that option, but it looks like I have to correspond with them for at least 2 weeks ;)
https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/67#issuecomment-998270714

stuarts0011, 2022-06-23:
> #67 (comment)

They did it for me

Santanachia, 2022-06-23:
another day and another answer - technically they have the capability, but "There is no senior engineer approval for such a downgrade."

Santanachia, 2022-06-24:
Another email, this time from a senior person.
They stated that for security reasons they would not downgrade the version and closed my request.

closed 2022-06-24 by Santanachia

Pvlerick, 2022-12-26:
According to this: https://www.solaredge.com/aus/setapp-inverters-firmware , Modbus should now work over WiFi:

> Modbus TCP over secured Wi-Fi is now officially supported for residential inverters