binsentsu/home-assistant-solaredge-modbus #92: Overriding deprecated methods on an instance of NumberEntity since HA 2022.7.0

pascalwinters, 2022-07-04:
```
2022-07-04 07:48:09 WARNING (MainThread) [homeassistant.components.number] custom_components.solaredge_modbus.number::SolarEdgeNumber is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
```

Cl3tus, 2022-07-06:
Same, subscribing.

rkuijer, 2022-07-12:
Related to :
https://github.com/home-assistant/core/issues/45658
Info:
https://www.home-assistant.io/integrations/number

Mosibi, 2022-07-16:
I see the same message in my logs

SalazarPellicer, 2022-07-16:
Lo mismo en los mios

Remko76, 2022-07-21:
I am afraid this integration is not really being maintained anymore, so the question is what do we do? I am not experienced yet in coding for HASS and do not really know where to start. Maybe someone here has an idea on how to modify this integration to make sure we do not run into issues after the 2022.10 upgrade? I think the required changes should not be too difficult, but I really have no clue how to start.

Remko76, 2022-07-21:
OK, dug around a bit and found what needed to be changed. I created a branch called "Patch1" and a pull request for that, which incoporates the changes required to get rid of the warning messages. You can find my changes here:
https://github.com/Remko76/home-assistant-solaredge-modbus/tree/patch-1/custom_components/solaredge_modbus

Easiest is to simply edit the file in your custom_components folder and restart Homeassistant. Now everything works for me again.

skl283, 2022-07-28:
> I am afraid this integration is not really being maintained anymore, so the question is what do we do? I am not experienced yet in coding for HASS and do not really know where to start. Maybe [...]

I agree! This Integration seems to not actively maintained! Perhaps https://github.com/WillCodeForCats/solaredge-modbus-multi is a good alternative? It is under heavy development and actively supported. Also it should support all that is supported by this Integration.

Whats you opinion?

sash

HyperActiveJ, 2022-07-29:
> I agree! This Integration seems to not actively maintained! Perhaps https://github.com/WillCodeForCats/solaredge-modbus-multi is a good alternative? It is under heavy development and actively [...]

I agree, ping @willcodeforcats and ask them how they feel about being the point contact.

WillCodeForCats, 2022-07-29:
I got a notice for the mention so I figured I'd reply.

I was using this integration and for while tried to contribute, but I was very new to all this at the time and pull requests for multiple inverter support weren't going anywhere (I suppose it's a less common need) so I gave up. Back then I didn't know anything about Home Assistant development, but in time I wrote my own integration (tekmar-482) then applied those lessons to my solaredge-modbus-multi integration. Although I had started a fork first, solaredge-modbus-multi is its own thing now with almost nothing in common.

It does not have the battery write command stuff, but it does have everything else plus a significant number of other improvements. There's too many differences to list so I'd recommend looking at the release history. Of course you can always try it to compare, plus I use it personally on my home system so I often develop improvements from my own observations, and of course keeping up with Home Assistant changes.

HyperActiveJ, 2022-07-29:
> I got a notice for the mention so I figured I'd reply.
>
> I was using this integration and for while tried to contribute, but I was very new to all this at the time and pull requests for multiple [...]

@WillCodeForCats  discounting missing the battery commands, is there any reason why someone would want to stick to this repo over your improved version?

Remko76, 2022-07-29:
@WillCodeForCats : you have done a fantastic job. I changed to your fork of the integration today and it is so much easier to set up and it works flawlessly!! Thank you very much for your work, I'll stick to it now..;)

skl283, 2022-07-29:
> It does not have the battery write command stuff

i'm switched also today - everything works great. Thank you for your great job! Are you planning to add the battery stuff to set the command like in this extension?

pascalwinters, 2022-08-03:
Fixed in v1.6.0

closed 2022-08-03 by pascalwinters

johanschelin, 2022-10-11:
Nice, tested and can confirm that this solves the issue, hopefully it will be merged, or someone else can take over maintaining the integration.

https://github.com/Remko76/home-assistant-solaredge-modbus/tree/patch-1/custom_components/solaredge_modbus