binsentsu/home-assistant-solaredge-modbus #97: Solaredge_status

geekypenguin, 2022-08-17:
Is it possible to change how the status is displayed?

I.e, instead of numbers 1-8 we get the status in human readable form

| Parameter              | Value | Description               |
|------------------------|-------|---------------------------|
| I_STATUS_OFF           | 1     | Off                       |
| I_STATUS_SLEEPING      | 2     | Sleeping - Night Mode     |
| I_STATUS_STARTING      | 3     | Grid Monitoring/Wake-up   |
| I_STATUS_MPPT          | 4     | Inverter on and producing |
| I_STATUS_THROTTLED     | 5     | Production Curtailed      |
| I_STATUS_SHUTTING_DOWN | 6     | Shutting Down             |
| I_STATUS_FAULT         | 7     | Fault                     |
| I_STATUS_STANDBY       | 8     | Maintenance               |

ChristophCaina, 2022-08-19:
Hi, the Readable status of the Inverter will be reflected in the attribute "status_description"

[image]

The easiest way - and that's what I am doing is to create a template sensor that will translate the status into a text:
```
# SolarEdge Inverter Status in Readable Text
    solaredge_status_text:
      friendly_name: "Solaredge status text"
      value_template: >-
        {% if is_state('sensor.solaredge_modbus_status', '1') %}
          Aus
        {% elif is_state('sensor.solaredge_modbus_status', '2') %}
          Ruhezustand
        {% elif is_state('sensor.solaredge_modbus_status', '3') %}
        fährt hoch
        {% elif is_state('sensor.solaredge_modbus_status', '4') %}
          Produzierend
        {% elif is_state('sensor.solaredge_modbus_status', '5') %}
          Gedrosselt
        {% elif is_state('sensor.solaredge_modbus_status', '6') %}
          fährt herunter
        {% elif is_state('sensor.solaredge_modbus_status', '7') %}
          Fehler
        {% elif is_state('sensor.solaredge_modbus_status', '8') %}
          Wartung
        {% else %}
          Unbekannt
        {% endif %}
        ```
        
For me this is the best option, because I can also translate the status text into german.
I would also need to do that if the sensor would directly display the status + I would have to consider the exact status text in my template.

You could create a template which would set the attribute status_description as state
for example like this:

```
template:
  - sensor:
      - name: inverter_status_text
        state: "{{ state_attr(''sensor.solaredge_modbus_status', 'status_description') }}"
```

geekypenguin, 2022-08-20:
Thanks, thats a much better idea.

Though I had to modify the code slightly to work. For anyone else who wants to replicate this, the second code block should read

`template:
  -sensor:
    -name: inverter_status_text
      state: "{{ state_attr('sensor.solaredge_status', 'status_description') }}"`

closed 2022-08-20 by geekypenguin