binsentsu/home-assistant-solaredge-modbus #99: Negative value on "number.solaredge_export_control_site_limit"

iNaiks, 2022-08-30:
Hi, I've recently write this problem on other issue. I write on a new issue for better search.

In my house the voltage of the grid is very high, I only can reduce using a little amount of energy of the grid. I tried to add negative value to number.solaredge.export_control_site_limit (in settapp I can add negative values on export control limit) for recive always 100~300w from the grid. But I can not set negative numbers because is out outside range. 

Question:
Is there any way to change the range?
Can I do or I need to wait to some update ?

(I very frustrated with this, because if the voltage of my house is more than 250v the inverter stop working and need 5-10 minutes to go working again, and if the voltage still over 250 it reboots again and again. Normally there are 10-60 reboots on a day... 😭.

Thanks
Nico

[image]
[image]

tias2001, 2022-08-30:
To me this sounds like the grid protection is tripping on your inverter as
per below
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.solaredge.com/sites/default/files/viewing_grid_protection_values.pdf&ved=2ahUKEwjX7bfGzu75AhVR4HMBHdRiAZUQFnoECB0QAQ&usg=AOvVaw3eLNUB6x60VbhKm58gmtxm

However I suggest you contact your installer to get them to fix it as it
sounds like it's not setup correctly

iNaiks, 2022-08-30:
The problem is on the grid. 
But my company of power is very slow to solve it. 

I only need to set negative numbers on export limit. 
There is any way?

mpredfearn, 2022-08-30:
The documentation for the export control site limit register
[image]
says that it is a float32 with value 0 - max site power - so you cannot set a negative value there by default.

However, I noticed that bit 11 of the "export control mode" register controls whether a negative value is allowable. 
[image]
Perhaps, if you modified the code to set bit 11 of that register, you could then set a negative export limit to force a constant import?

tias2001, 2022-08-30:
I don't think that the export limit can be used in the way you want it to

iNaiks, 2022-08-30:
Ohh very thanks for the info. 
But in the settapp I can set a negative number. 
later I will post the photo. 

Any idea how to do it?
Actually I’m trying to set in production limitation with  “usage of the house -  input number with the power import” but when my usage of the house changes a lot, the production limitation limits too much or too less.

mpredfearn, 2022-08-30:
Yeah I think if you set bit 11 in the export control mode register, you would be able to set a negative limit.

However, it sounds like you're trying to achieve real time control of the inverter here - I don't think that that is going to work at all well - the latencies will be too high, and if home assistant goes offline for any reason you could leave the inverter in a bad state.

I am not sure what you are trying to achieve is really possible - you need to get your grid connection fixed to prevent the inverter tripping. Sounds like you need the installer / grid operator in to sort it.

RobinXe, 2023-01-10:
I hope you got this fixed already but, if not, there are a couple of things you could try.

I'm assuming that you're in Europe, where the tolerance for grid voltage is 230 V -6% +10%, so 216 - 253 V, so your supply sounds like it's operating toward the upper limit. It would make sense that the grid protection settings on your inverter match the values of the full potential potential range (see what I did there?!). If you don't have administrative access to your inverter's settings to make the adjustments, then contact your installer, or SolarEdge directly if they won't/can't help.

An additional option is to enable reactive power control on the inverter, to modify the power factor and help drag the grid voltage back from the limits of tolerance. It's a somewhat complicated topic, but there's a fairly good explanation (from the Australian market) [HERE](https://pure-electric.com.au/news/how-correct-reactive-power-settings-your-inverter-can-increase-your-solar-revenue), and details of how to implement the system on SolarEdge inverters [HERE](https://knowledge-center.solaredge.com/sites/kc/files/application_note_power_control_configuration.pdf?fbclid=IwAR10B7uCx2yyuvnNM7Gu48KgGHHjyUd4nbgLvTPvIqX1SdOKIlkdrWrRqeI).