binsentsu/home-assistant-solaredge-modbus #100: Causes Unresponsive HA UI

Meatballs1, 2022-08-31:
If connectivity is lost to the solaredge (replicate with `sudo iptables -I OUTPUT -d 192.168.0.x/32 -j DROP`) the UI becomes very unresponsive. 

I have modified my DEFAULT_SCAN_INTERVAL to 5s which probably causes the issue due to the timeout period being greater. It doesn't seem to affect a scan interval of 30s (or isn't that noticeable). 

The following errors seen in the log (as expected):

```
2022-08-30 21:00:17 ERROR (MainThread) [pymodbus.client.sync] Connection to (solaredge-inverter-wired.xxx.xxx, 1502) failed: timed out
2022-08-30 21:00:17 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 228, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 552, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=1, address=40071, count=38)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 215, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 106, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(solaredge-inverter-wired.xxx.xxx:1502)]
2022-08-30 21:00:20 ERROR (MainThread) [pymodbus.client.sync] Connection to (solaredge-inverter-wired.xxx.xxx, 1502) failed: timed out
```

I suspect it is blocking the connection for a standard timeout period and probably using up the thread pool but I dont really know a lot about HA internals.

Modifying __init__.py with the following seems to relieve the issue:

```
self._client = ModbusTcpClient(host=host, port=port, timeout=(scan_interval-1))
```

There is also  retries=0, reconnect_delay=0 that could help but didn't seem to be required.

donnie-darko, 2022-12-14:
Thanks for reporting this issue @Meatballs1 because I also experience this unresponsive UI when lost connection to the solaredge since forever but never have the chance to open an issue. until now. and thanks for the workaround, too! 
cheers

svewag, 2023-08-04:
I also can confirm that this works for me

closed 2023-09-12 by binsentsu

alexdelprete, 2023-09-16:
> which probably causes the issue due to the timeout period being greater

the default timeout for ModbusClient (ModbusTCPClient derives from it) is 3s: https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.ModbusBaseClient

[image]

So you had a scan_interval of 2s greater than default timeout. After that code change you have 1s.

PR #183 (2023-09-12): make timeout dependent on scan interval
In order to prevent an unresponsive UI in Home Assistant in case of connectivity issues of the modbus server, the client now uses not the default timeout but a timeout dependent on the selected scan interval.

This fixes #100
commit 6032d26f (2023-09-12): make timeout dependent on scan interval

In order to prevent an unresponsive UI in Home Assistant in case of connectivity issues of the modbus server, the client now uses not the default timeout but a timeout dependent on the selected scan interval.

This fixes #100