binsentsu/home-assistant-solaredge-modbus #102: Allow changing of config from Integrations page

ryanm101, 2022-09-06:
Once i set mine up i was on DHCP for my IP which fell within my automatic range. When i set a static reservation I had to delete and recreate the integration. 

I'm currently waiting on batteries which have not arrived yet but I'm going to have to delete and readd the integration again when they arrive as well. 

Allowing the full config to be changed without deleting and readding the integration

gferen, 2022-09-23:
I would be also grateful to be able to change integration settings, instead of removing and adding back. I just need to change setting not to keep connection alive.

ChristophCaina, 2022-10-20:
yes, would be great.
Right now, you need to define the amount of meters, batteries, and so on during the setup.

Else, the status of all sensors would be "unknown".

If something on your setup changes, it would be great to be able to change these type of things within the UI...
- Inverters IP address
- Connection Port
- Amount of connected Meters, Batteries, etc.

ryanm101, 2022-10-20:
I think `Amount of connected Meters, Batteries, etc` should be 'detectable' based on values returned in the common block.

bramvandenbussche, 2023-02-08:
+1 on this fine suggestion. Having to recreate it to fiddle with the options is not fun.

rehanson, 2023-02-21:
I would like this feature as well. I had to set up a new solaredge modbus device in homeassistant after changing the inverter to a new IP address, and it really complicated my Energy Dashboard history.

ChristophCaina, 2023-02-21:
you can change the IP Address of the inverter by just changing it within the /config/.storage/core.config_entries
there, you should find the configuration for the solaredge_modbus component and there, you can change the ip address (host).
That should help if it's only about a small change (like port, ip or to add / remove a battery / meter)

I know, it is a bit more complicated, but would allow you to edit the changes without removing the integration at all.

So maybe, to consider as a workaround :)

rehanson, 2023-02-21:
My understanding was that messing with the core.config_entries file wasn't really recommended. I don't want to cause any problems by manually editing it.

fredmoro, 2025-01-15:
I encountered the above problem: had to change the IP address.. On solaredge integration page I've clicked "Add entry" ,then I entered the new address (here the meters and battery related options can be checked too.) It started to work, after reloading the config. Later I've deleted the old entry with wrong IP address. The configured dashboard, energy, and other things seems recovered and working again... Make a backup, before You try it, in my case it was a solution, instead of editing the config (I found this topic AFTER I've done the above procedure :D )
p.s.:
After recovered the communication with the solaredge inverter, the energy card's solar production shows a false, negative data to today's solar production - that I'll have to remove from the database manually...