binsentsu/home-assistant-solaredge-modbus #104: Cannot add battery control

silverrick88, 2022-09-12:
Think this may be more of a solaredge issue but is the any way to add remote control for the newer inverters without a screen? I can only see options for MSC and ToU. SE4000H specifically for me.

ryanm101, 2022-10-20:
I'd not got as far as that yet but it's on my todo list too, but i'd noticed the same thing.. I cant seem to find a working link to the ToU docs either.

ryanm101, 2022-12-05:
@silverrick88 I've just set this up.

```yaml
alias: Charge Battery From Grid
mode: single
trigger:
  - platform: state
    entity_id:
      - select.household_energy_daily
    to: offpeak
  - platform: state
    entity_id:
      - select.solaredge_storage_remote_command_mode
    to: Maximize self consumption
  - platform: time
    at: "06:30:00"
condition:
  - condition: and
    conditions:
      - condition: state
        entity_id: select.household_energy_daily
        state: offpeak
      - type: is_value
        condition: device
        device_id: 758241f74124df3fe7af56b6d84e1410
        entity_id: sensor.solaredge_battery1_state_of_charge
        domain: sensor
        below: 99
action:
  - service: select.select_option
    data:
      option: Charge from PV and AC
    target:
      entity_id: select.solaredge_storage_remote_command_mode
```
```yaml
alias: Charge Battery From PV Only
description: ""
mode: single
trigger:
  - platform: state
    entity_id:
      - select.household_energy_daily
    to: peak
action:
  - service: select.select_option
    data:
      option: Maximize self consumption
    target:
      entity_id: select.solaredge_storage_remote_command_mode
```

ryanm101, 2022-12-05:
If you look in `integrations` then click on the solaredge `entities` then in the search box search for `input` and `select`

ryanm101, 2023-01-22:
I changed the above ones to this as previously the remote command was timing out and because my default was max usage i was then draining offpeak and having to charge again..
Now I switch to charge at offpeak then once charged I switch the battery off and turn it on again at peak rate where i set back to max self usage
```
alias: Charge Solar Battery From Grid
mode: single
trigger:
  - platform: state
    entity_id:
      - select.household_energy_daily
    to: offpeak
condition:
  - condition: and
    conditions:
      - condition: state
        entity_id: select.household_energy_daily
        state: offpeak
      - type: is_value
        condition: device
        device_id: 758241f74124df3fe7af56b6d84e1410
        entity_id: sensor.solaredge_battery1_state_of_charge
        domain: sensor
        below: 100
action:
  - service: select.select_option
    data:
      option: Charge from PV and AC
    target:
      entity_id: select.solaredge_storage_default_mode
```

```
alias: Charge Solar Battery From PV Only
description: ""
mode: single
trigger:
  - platform: state
    entity_id:
      - select.household_energy_daily
    to: peak
action:
  - service: select.select_option
    data:
      option: Maximize self consumption
    target:
      entity_id: select.solaredge_storage_default_mode
```

```
alias: Disable Solar Battery
description: ""
mode: single
trigger:
  - platform: numeric_state
    entity_id: sensor.solaredge_battery1_state_of_charge
    above: 99
condition:
  - condition: state
    entity_id: select.household_energy_daily
    state: offpeak
action:
  - service: select.select_option
    data:
      option: "Off"
    target:
      entity_id: select.solaredge_storage_default_mode
```