binsentsu/home-assistant-solaredge-modbus #109: multiple devices, not rs-485 chained support

vvuk, 2022-10-01:
I have two inverters that are not chained together via RS-485.  Previously they were both communicating with a Tesla Zigbee gateway box; I've since connected them both via ethernet.  They have independent IP addresses, and for modbus purposes, both think they are modbus ID 1.

Do I need to run a second instance of the integration?  Or am I missing something with the inverter setup?  Thanks!

ChristophCaina, 2022-10-17:
Hi,
as far as I know, the integration does only communicate with one IP.
If your Inverters have different IPs, then you need one integration for each Inverter.

But I would recommend to consider the following setup (if possible)
Connect both inverters to each other.
This would allow them to share information between each other.
I think, this would also allow you to have both inverters combined in one Dashboard on SolarEdge Monitoring Platform (if they can be connected to each other).

In this case, maybe the multi-inverter fork of this integration might be relevant, since I don't know if this integration does support multiple inverters.

jopvanraaij, 2023-07-10:
I guess it is resolved by changes in HA in the meantime (using 2023.7.1). I added 2 different devices with a different IP each. Using the 'ADD ENTRY' in the screenshot _did not work_ though. I had to add the integration again, for the second device, from the main integrations-page, as if it was adding the whole integration again.

[image]