binsentsu/home-assistant-solaredge-modbus #110: Component wont load when battery is empty

ryanm101, 2022-10-16:
```
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 251, in read_modbus_data
    and self.read_modbus_data_battery1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 773, in read_modbus_data_battery1
    return self.read_modbus_data_battery("battery1_", 0xE100)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 858, in read_modbus_data_battery
    battery_SoC = validate(decoder.decode_32bit_float(), ">", 0.1)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 130, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0.0 failed validation (>0.1)
```

tias2001, 2022-10-16:
Yes I get that too

ryanm101, 2022-10-16:
I just changed line 858 to:
`battery_SoC = decoder.DECODE_32bit_float()`
and reloaded it

marria05, 2022-10-19:
I had my inverter and battery installed this afternoon and ran into the same issue. Thanks for the quick fix @ryanm101. 

However, having not done much tinkering inside the code custom components up to now, I didn't realise you needed to do a full reload of HA, not just a integration itself. 

Took me a good 15 mins to realise why my code changes didn't seem to take. Hopefully someone in the same predicament might see this before tearing their hair out.

ryanm101, 2022-10-20:
@marria05 sorry about that i've edited my comment to make it more clear

ryanm101, 2022-10-20:
[image]

ChristophCaina, 2022-10-23:
@ryanm101 I think, that was related to the comment in the PR that you don't validate against a negative value... ?!
I hope, the PR does work as intended... but it should be verified by someone who has a battery ... ?!

ryanm101, 2022-10-23:
I think you are correct. I do think though the check should be `>= 0.0` and not `>0.0` as imo `0.0` is a valid non-negative result.

ryanm101, 2022-10-23:
@ChristophCaina I have a battery. I've removed my neg entries from the db and am running the current master (with my refactor) for testing and it looks good so far.

ryanm101, 2022-10-25:
Still getting negative entries.. strange

ryanm101, 2022-10-26:
```
12342755|sensor.solaredge_battery1_state_of_charge|-3.40282346638529e+38||||2022-10-24 19:11:28.504682|12332559|751037|01GG5PGC7RNNTTQHS679V62ZV3|||0
12461840|sensor.solaredge_battery1_state_of_charge|-3.40282346638529e+38||||2022-10-25 05:12:06.554045|12342935|751037|01GG6RW5MTBT0QAZAW51TEY1R6|||0
12584023|sensor.solaredge_battery1_state_of_charge|-3.40282346638529e+38||||2022-10-25 12:19:35.491396|12582575|751037|01GG7HAXE30BGT6T6Z2XZRVHDR|||0
12697795|sensor.solaredge_battery1_state_of_charge|-3.40282346638529e+38||||2022-10-25 18:55:39.939556|12696432|751037|01GG8804X3TSQ07YYZMZJQE33M|||0
```

ChristophCaina, 2022-10-26:
can you show me the graphs?
Do you have an idea where this negative values could come from?

Maybe, another check somewhere else in the code is required?

I've just changed the part that was suggested here and created an pr from it... I haven't checked anywhere else

ryanm101, 2022-10-26:
Graph is as above..
I cleared the old values and got a nomal graph last night. Today I have the same neg issue.

I think you are correct there must be somewhere else that needs the validation

ryanm101, 2022-10-26:
```
custom_components/solaredge_modbus/const.py
250:    "BATTERY1_SOC": ["Battery1 State of Charge", "battery1_state_of_charge", "%", "mdi:battery-high"],
265:    "BATTERY2_SOC": ["Battery2 State of Charge", "battery2_state_of_charge", "%", "mdi:battery-high"],

custom_components/solaredge_modbus/__init__.py
858:            battery_SoC = validate(decoder.decode_32bit_float(), ">", 0.0)
859:            battery_SoC = validate(battery_SoC, "<", 101)
871:            self.data[battery_prefix + 'state_of_charge'] = round(battery_SoC, 0)
```

ok strange i cant see how it could be getting set while. negative.

ryanm101, 2022-10-26:
Interestingly all the rogue values are the same. `-3.40282346638529e+38`

ryanm101, 2022-10-26:
hold on i might be an idiot. 

I think i am. I thought i'd overwriten the install with my code but looks like i've screwed that up with the upload. i may be using the unvalidated check :/

Sorry for the confusion.

ryanm101, 2022-10-26:
I'll verify all is good in the world tomorrow when my battery has been drained.

ryanm101, 2022-10-28:
```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:130
Integration: SolarEdge Modbus (documentation)
First occurred: 08:57:18 (9 occurrences)
Last logged: 09:01:19

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 251, in read_modbus_data
    and self.read_modbus_data_battery1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 773, in read_modbus_data_battery1
    return self.read_modbus_data_battery("battery1_", 0xE100)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 858, in read_modbus_data_battery
    battery_SoC = validate(decoder.decode_32bit_float(), ">", 0.0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 130, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0.0 failed validation (>0.0)
```

ryanm101, 2022-10-28:
per previous comment needs to be `>=`

ChristophCaina, 2022-10-28:
Hi, can you please Check, of this is also valid for #113

closed 2023-01-16 by ryanm101