binsentsu/home-assistant-solaredge-modbus #111: After last HA update addon stoped working

BooBoss, 2022-10-19:
As title says. From 4 days addon is not getting any data from inverter

[image]

All values says "unknown".

I've reinstalled addon but still the same.

BooBoss, 2022-10-20:
I found in logs this:

`Ten błąd pochodzi z niestandardowej integracji.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:233
Integration: SolarEdge Modbus (documentation)
First occurred: 19 października 2022 18:35:06 (1854 occurrences)
Last logged: 10:01:43

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 246, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 570, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=self._address, address=40071, count=38)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 233, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 106, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.99:1502)]
`

And also this:

`Ten błąd pochodzi z niestandardowej integracji.

Logger: pymodbus.client.sync
Source: custom_components/solaredge_modbus/__init__.py:233
Integration: SolarEdge Modbus (documentation)
First occurred: 19 października 2022 18:35:06 (1854 occurrences)
Last logged: 10:01:43

Connection to (192.168.0.99, 1502) failed: [Errno 111] Connection refused
Connection to (192.168.0.99, 1502) failed: timed out
`

Also I checked again on inverter that TCP modbus is enabled so I don't know why it refuses connection.

ChristophCaina, 2022-10-20:
OK...
At least, that's some kind of information.
Try to Change the Port from 1502 to 502 - I Had to do this after Connecting a modbus Proxy between the Inverter and HomeAssistant.

It might be, that the Inverter is changing the Port when a new Connection will be detected

BooBoss, 2022-10-20:
Still nothing. After changing port to 502 all entities still has no values and found this in logs:

```
Ten błąd pochodzi z niestandardowej integracji.

Logger: pymodbus.client.sync
Source: custom_components/solaredge_modbus/__init__.py:217
Integration: SolarEdge Modbus (documentation)
First occurred: 19 października 2022 18:34:35 (3 occurrences)
Last logged: 10:46:03

Connection to (192.168.0.99, 1502) failed: [Errno 111] Connection refused
Connection to (192.168.0.99, 502) failed: [Errno 111] Connection refused
```

Funny thing is he's trying to connect both on ports 1502 and 502.

BooBoss, 2022-10-20:
Maybe the modbus address should be changed?

[image]

superharaldsson, 2022-10-20:
Same for me. :/

BooBoss, 2022-10-20:
> Same for me. :/

From how long?

BooBoss, 2022-10-20:
> Since the last update (2022.10.4)

So exactly same as me. In other words it has to have something to do with update itself. Did you tried by any chance restart whole inverter? Like power it down and start again? I came into this idea today but I'm in work ATM.

superharaldsson, 2022-10-20:
At least I think so.  Yes, I tried that yesterday.  Have also tried changing the modbus address but no difference.

BooBoss, 2022-10-20:
So... here are firmwares versions:

[image]

Also found errors in inverters like this, however last error was after it stoped working. Second error from Oct 16 might be around when it stoped working. However it's about some voltage - no communication:

[image]

Also restarted inverter today - no luck.

ChristophCaina, 2022-10-20:
ok, the cpu version is the same as it is on my setup... the DSP1 and DSP2 versions are a bit older... but honestly, I don't know if this might be related.

I would suggest to change the setup:

if you're running HomeAssistant Supervised (or HassOS) - install the Addon "Modbus-Proxy"

```
Modbus Proxy Repository
Julian Jost <julian@jost-online.eu>
https://github.com/Akulatraxas/ha-modbusproxy
```

Connect this to your inverter.
The setup is pretty simple:
```
upstreamhost: <IP from your inverter>
upstreamport: 502
listenport: 502
```
upstreamport can be 1502 or 502 and needs to be tested.

The addon will log a successfull or failed connection.
This is to make sure, that it's not the integration or HomeAssistant itself which is causing the issue.

Then, configure your SolarEdge Modbus integration to the IP of your HomeAssistant installation - with the "listenport"...

The benefit of this setup:
You could also connect other devices that might require information from your inverter to the modbus-proxy.
I've done this to connect HomeAssistant and my Wallbox with the Inverter - because SolarEdge does only allow one active connection at the same time.

If something else does connect to the inverter - while HomeAssistant is offline, it will prevent HA / the integration to establish the connection.

BooBoss, 2022-10-20:
OK - I can't check it right now but I will check it and report back. Tomorrow at latest. Thanks for your feedback so far.

binsentsu, 2022-10-22:
Doesn't seem to me like a general issue. Running myself on 2022.10.5 without the issue

BooBoss, 2022-10-22:
I am running HA on HassIO but I can't find "Modbus-Proxy" in addons repository. Where did you found it?

ChristophCaina, 2022-10-22:
I will Share the repository later - but I've also a fork in my repositories... Holen Sie sich Outlook für Android

PrimeUFO, 2022-10-23:
Hi guys,

I have the same problem, I tried almost everything, I change the port (6623) and the port stays open until  I reintegrate HA and then will stop responding.

 It's very weird because is no firewall on local network and the invertor should only send data. I will try with 502, will see if that will make any difference.

Thanks

ChristophCaina, 2022-10-23:
here's the repo for the modbus proxy
https://github.com/Akulatraxas/ha-modbusproxy

Blom1974, 2022-10-26:
Got the same problem. All sensors unknown.

ChristophCaina, 2022-10-26:
Hi @Blom1974 - it would be helpful to get actual logs ...
There are several reasos why the integration could get "unknown sensor states...

without any kind of information to start with, it's hard to support

psqf, 2022-10-26:
Hi, here my log with the same problem:

Questo errore ha avuto origine da un'integrazione personalizzata.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:233
Integration: SolarEdge Modbus (documentation)
First occurred: 25 ottobre 2022 19:23:16 (3199 occurrences)
Last logged: 22:02:17

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 246, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 570, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=self._address, address=40071, count=38)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 233, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.65:1502)]

ChristophCaina, 2022-10-27:
@psqf :
The log indicating an issue with your connection:
```
ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.65:1502)]
```
and
```
Connection to (192.168.1.65, 1502) failed: timed out
Connection to (192.168.1.65, 1502) failed: [Errno 111] Connection refused
```

1. how many meters / batteries are installed, and how many have you selected in the configuratioin of the integration?
- Did it already work?

2. You can try to switch to Port 502 - SolarEdge is either using 502 OR 1502... both can be possible
3. Is your connection Wifi, or Ethernet?
If Wifi, please check the Firmware Version on your inverter and check if it is one of the versions mentioned in the release nots:
https://www.solaredge.com/de/setapp-inverters-firmware --> for inverters without display (SetApp)
https://www.solaredge.com/de/service/firmware --> for inverters with display

psqf, 2022-10-27:
Thanks: I have a 3phase inverter without display , but from online monitoring I found CPU 4.15.119 firmware. I don't know, these inverters could self-upgrade? I try 502 port too but keeps refuse connection (if I ping it's OK). My connection is Wifi. So next days I will try to re-enable modbus connection (maybe auto-off?) or I'll need a rs485 >>IP converter.

ChristophCaina, 2022-10-27:
a 169 IP is most probably an internal ip.
Does your Inverter has a static IP or is it set to DHCP?

It might be, that it did not get a new IP lease from the DHCP - and therefore switched to a default 169.x. network...
This COULD also explain why you can't connect to the inverter.

If it is set to DHCP, I would consider to asign a static IP to it - and make sure, that the Wifi Connection is working.
If not, I would always prefer a wired connection - if possible...

BooBoss, 2022-10-28:
> In web monitoring site (display replica), the IP is reported is 169.XXX and not 192.XXX it's OK?

This address doesn't make any sense. 169.XXX addresses are from auto-config. In other words if you have ethernet adapter set to DHCP but there is no any DHCP server on the network then adapter will pickup random 169 address. This is used in small networks without any DHCP server so devices can communicate between each other on TCP/IP without address assigned by DHCP server.

psqf, 2022-10-28:
The inverter is connected via wifi and I can see data on APP/monitoring site so it is connected to internet. The router is DHCP but I make static the inverter 192.168.1.x. I can see and ping from Home assistant server, so it's strange the data reported. My opinion is that someone have upgraded the firmware and disabled Modbus connection :(

ChristophCaina, 2022-10-28:
Solaredge had removed Support for modbus via WiFi.
If your Firmware is below the one given in the Release notes, you need to Open a Ticket with SE.

BooBoss, 2022-10-28:
I don't know if I had any OTA update or not but for me TCP ModBus seems to be still available (however not working):

[image]

And the firmware versions looks like this:

[image]

psqf, 2022-10-28:
But you connect via ethernet or wifi? I think I should use a repeater with LAN port to re-establish connection to HA

BooBoss, 2022-10-28:
Via WIFI - I don't even think there is ethernet card there. If there is I can put a cable there if that would fix the problem. Maybe I will just take off front cover to look for ethernet but I don't see it anywere on the body.

psqf, 2022-10-28:
My version triphase 7KW is internal, look at manual

ChristophCaina, 2022-10-28:
Usually, these Inverters should have Ethernet.
WiFi isn't the preferred Method for such devices.

The Release notes are linked a few Comments above.
I think with 4.19.115 they had issues with WiFi.

Just try to do a Test with Ethernet. If this is working & you prefer wifi, Open a Ticket with Solar Edge

BooBoss, 2022-10-28:
I can go with ethernet. Just put cable there. I always prefer cable over WIFI wherever possible. Think is I don't see ethernet socket anywhere on the body.

ChristophCaina, 2022-10-28:
It's probably somewhere Inside.
Also on my Inverter, the Ethernet cable seems to Go into the housing of the Inverter

BooBoss, 2022-10-28:
OK - I will just take out the cover and look for it. It's probably somewhere inside on the board itself.

psqf, 2022-10-28:
solaredge website is full of product description, the comm board has both LAN and wifi antenna connector. Pay attention to wait the time for discharge capacitor because it's very dangerous!

BooBoss, 2022-10-28:
So this is a mainboard of my iverter. That's a real photo taken by me just 10 minutes ago:

[image]

This blue area is WIFI module while red circle is antena connector. 

I looked very closely at the mainboard and I didn't found ethernet connector anywhere on the board. Of course I might be blind but I don't see it anywhere.

(And I didn't switched it off before but fortunatelly I'm still alive xD)

ChristophCaina, 2022-10-28:
I'm not 100% sure - but I had a quick search and found a video, where this area was indicated as ethernet - port.

And yes, while looking at the layout, I would consider this as Layout for an ethernet-connector :)
[image]

BooBoss, 2022-10-29:
So in other words I don't have ethernet socket soldered on the mainboard - great ;-)

psqf, 2022-10-29:
I th8nk it is on botton side ef board, you can reach if you see from bottom or if you remove board with 4 screw

BooBoss, 2022-10-29:
> I th8nk it is on botton side ef board, you can reach if you see from bottom or if you remove board with 4 screw

OK - I will check that at the evening when production stops. Don't want to interrupt energy production during day.

sarnassj, 2022-10-30:
So... Me and my friend (with exactly the same Solar edge and HA installation) have the same problem. Integration stop to works 24 October. I even don't update HA. Still have 2022.9.6 version. I get the same error:

`(pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.68.59:1502)) `

I think SolarEdge get OTA update,  and something was changed in ModBus Settings. But I need to confirm that. Unfortunetlly f***ing SolarEdge block this settings, and it's available only for certificated installatiors so I need to contact with mine.

But the conclusion is: I didn't make an update of HA so I think the error is connected more with SolarEdge device then with the HA itself.

ChristophCaina, 2022-10-30:
as mentioned in previous comments here:
1. SolarEdge does OTA updates on the Inverters (there's no indication when this happens)...
2. With some of the Firmware versions, they have disabled ModbusTCP over WiFi - after several complains, this was fixed with the recent firmware version, mentioned in the release notes (link somewhere above).
3. If possible, I would recommend a wired connection over wifi ...
4. you can check the inverters firmware version and ModbusTCP settings with SetApp (if the inverter does not have a Display)...
How the Inverter can be used with setapp should be described on the websites of SolarEdge
5. if the Firmware Version has changed on the inverter - or is one of the "affected" ones, you can also open a ticket with SolarEdge to update to the latest firmware, where issues with wifi were fixed.

--- Probably, this "issue" should be moved to discussions...
@binsentsu - could you enable the discussions section?
I think, there could be a benefit in cases like this to have a discussion instead of issues :)

binsentsu, 2022-10-30:
@ChristophCaina Discussions should be available now

BooBoss, 2022-10-30:
> I think SolarEdge get OTA update, and something was changed in ModBus Settings. But I need to confirm that. Unfortunetlly f***ing SolarEdge block this settings, and it's available only for certificated installatiors so I need to contact with mine.

Are you sure about that? There's this switch [P 0 1] under case. Switch it to P for a second and inverter will start up it's own Access Point. Then you can connect to it's WIFI from mobile and go to address http://172.16.0.1 to see all settings. At least this is how it works in my case.

BooBoss, 2022-10-30:
> I th8nk it is on botton side ef board, you can reach if you see from bottom or if you remove board with 4 screw

I must admit - I am completelly blind IDIOT!

[image]

I have no idea how I didn't see that at first time.

Anyway I will connect ethernet just by putting cable over floor for a moment and see if it works. Then report back of course.

BooBoss, 2022-10-30:
CONFIRMED WORKING OVER ETHERNET!!!

So seems like there was firmware upgrade. Works on ethernet anyway. 

Thanks to anyone involved in this topic!!!

psqf, 2022-10-30:
thanks for report!

Blom1974, 2022-10-31:
SolarEdge updated my firmware to 4.16.23 which should be supported wirelessly but still not working.
Could someone elaborate on what secured/unsecured actually means in this context.
I am unfortunately not able to connect via Ethernet.

From changelog:

Modbus TCP over secured Wi-Fi is now officially supported for residential inverters
 - Modbus TCP over unsecured Wi-Fi is blocked
 - Modbus TCP over Wi-Fi for commercial inverters is blocked

superharaldsson, 2022-10-31:
For me it was the Solar Edge update that caused it. Contacted Solar Edge by email and they updated immediately and now it works.

Blom1974, 2022-10-31:
> For me it was the Solar Edge update that caused it. Contacted Solar Edge by email and they updated immediately and now it works.

Do you know which firmware you are running?

superharaldsson, 2022-10-31:
4.16.23

Blom1974, 2022-10-31:
> 4.16.23

Same as me then. Will do some testing when I get home tonight.

BooBoss, 2022-10-31:
My previous message about synchronisators was bullshit so I deleted that. I've contacted support. Just in case if someone doesn't know - if you power on inverter when it's dark outside synchronisators won't synchronise because they need electric power produced by sun. What's worse they won't even synchronise next day morning. What you have to do in such case is either restart inverter or force synchronisation from SetApp but DURING DAY! (when sun operates).

BooBoss, 2022-10-31:
I've noticed one stupid thing. When I connected ethernet finally and disconnected WIFI then I also lost ethernet connection as well. In other words I have to have both WIFI and ethernet operating to have ethernet working. Kinda ridicoulus.

ChristophCaina, 2022-11-01:
> I've noticed one stupid thing. When I connected ethernet finally and disconnected WIFI then I also lost ethernet connection as well. In other words I have to have both WIFI and ethernet operating to have ethernet working. Kinda ridicoulus.

That really sounds strange.
It should work either over wifi OR ethernet... 
But maybe, some other settings on your inverter might be relevant for this.

Unfortunately, I cannot provide any detailed information on how to setup the inverter in this case...
Maybe, SolarEdge Support can help in this case?

BooBoss, 2022-11-02:
> That really sounds strange. It should work either over wifi OR ethernet... But maybe, some other settings on your inverter might be relevant for this.
>
> Unfortunately, I cannot provide any detailed [...]

Honestly I won't even bother to check. It works with both adapters running so I will just leave it as it is.

BooBoss, 2022-11-05:
Sooo.... even tho I connected inverter by ethernet and it was working for few days now I have AGAIN - connection refused error. Same as on previous screenshots above.

miklosandras, 2022-11-05:
I have the same issue, reinstalled the integration not helped.

psqf, 2022-11-05:
:( . I suggest open a ticket on solaredge site, it is simple, if you report serial they control online in few days