binsentsu/home-assistant-solaredge-modbus #113: ValueError: Value 0 failed validation (>0)

ChristophCaina, 2022-10-23:
getting the following error in the Logs:
```
Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 247, in read_modbus_data
    and self.read_modbus_data_meter1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 259, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 423, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 130, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)
```

Checking the Source, it seems to be the meter export value?!

line 195 (194 - 198):
```
        try:
           update_result = self.read_modbus_data()
        except Exception as e:
           _LOGGER.exception("Error reading modbus data")
           update_result = False
```

line 247 (244 - 253):
```
    def read_modbus_data(self):
        return (
            self.read_modbus_data_inverter()
            and self.read_modbus_data_meter1()
            and self.read_modbus_data_meter2()
            and self.read_modbus_data_meter3()
            and self.read_modbus_data_storage()
            and self.read_modbus_data_battery1()
            and self.read_modbus_data_battery2()
        )
 ```
 
 line 423:
 ```
             exported = validate(self.calculate_value(exported, energywsf), ">", 0)
```

line 130:
```
    if not ops[comparison](value, against):
        raise ValueError(f"Value {value} failed validation ({comparison}{against})")
    return value
```

I am not quite sure - shouldn't be the validation something like
exported = validate(self.calculate_value(exported, energywsf), ">=", 0)
Also, I don't really know where the value 0 came from - probably, because there was an error reading the modbus-data?

ChristophCaina, 2022-10-28:
It should be checked, of other validations needs to be >= 0 instead of using >0

ryanm101, 2022-10-28:
https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/114 should fix this

closed 2022-10-28 by binsentsu