binsentsu/home-assistant-solaredge-modbus #121: Solar production

brontolo88, 2022-11-09:
Hi, with this integration I'm not able to see the solar production. I can see the power coming from the invert, from the batteries and from the grid in both directions (with + or - sign) but I can't see directly the power production. Is there a way to fix it?

mpredfearn, 2022-11-09:
https://github.com/binsentsu/home-assistant-solaredge-modbus/wiki/Using-Templated-Sensors-to-Calculate-Power-Flow-and-Energy

blowk, 2022-11-13:
During the night the sensor sensor.solaredge_modbus_ac_energy_kwh is also in creased when the battery is being charged from the grid.
How can you calculate the solar production only?

mpredfearn, 2022-11-14:
See the `energy_solar_generation` sensor template in he above link.

blowk, 2022-11-14:
Thanks.
I had a look, but I cannot find the sensor energy_battery_charged_solar.
Only have solaredge_modbus_battery_charged, but the battery can be charged from the grid at night.
Same with energy_battery_discharged. This can be energy from the grid.
I'm missing something?

closed 2022-11-30 by binsentsu