binsentsu/home-assistant-solaredge-modbus #123: No communication when battery is 0

enry86cami, 2022-11-17:
Hi all,
When the battery is 0 I noticed that the modbus connection stops working....
This is the log:

Error reading modbus data
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 251, in read_modbus_data
    and self.read_modbus_data_battery1()
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 773, in read_modbus_data_battery1
    return self.read_modbus_data_battery("battery1_", 0xE100)
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 858, in read_modbus_data_battery
    battery_SoC = validate(decoder.decode_32bit_float(), ">", 0.1)
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 130, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0.0 failed validation (>0.1)

Error reading modbus data
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 246, in read_modbus_data
    self.read_modbus_data_inverter()
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 570, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=self._address, address=40071, count=38)
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/__init__.py", line 233, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/srv/homeassistant/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/srv/homeassistant/lib/python3.9/site-packages/pymodbus/client/sync.py", line 106, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.30:1502)]

When battery is 0 no data, see below:
[image]

What could I check?

ryanm101, 2022-11-17:
Change and use master branch instead of the 1.6.0 and this is fixed

enry86cami, 2022-11-18:
> Change and use master branch instead of the 1.6.0 and this is fixed

sorry, what is master branch?

tias2001, 2022-11-18:
> ID: <binsentsu/home-assistant-solaredge-modbus/issues/123/1319677858@
> github.com>

Open hacs, open solaredge modbus, go to redownload, change version to
master and download

ryanm101, 2022-11-18:
as per @tias2001 comment. Then ignore HACs telling you there is a new version until it is greater than `1.6.0`

enry86cami, 2022-11-18:
thanks for your reply.
I downloaded the master version...
Now I'm waiting for battery goes emply....

Zumbug, 2022-11-19:
I am using master (1.4.0) and still getting an error when the battery is on 0%:

Error reading modbus data
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 195, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 251, in read_modbus_data
    and self.read_modbus_data_battery1()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 773, in read_modbus_data_battery1
    return self.read_modbus_data_battery("battery1_", 0xE100)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 858, in read_modbus_data_battery
    battery_SoC = validate(decoder.decode_32bit_float(), ">", 0.1)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 130, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value -3.4028234663852886e+38 failed validation (>0.1)

enry86cami, 2022-11-19:
The problem Is still present with the master version...

ryanm101, 2022-11-20:
strange my battery is empty at the moment and it works fine.

ryanm101, 2022-11-20:
have you tried removing and re-adding the integration?

binsentsu, 2022-11-30:
Code change wasn't pushed out as a release yet. Release now available (V1.7.0) @enry86cami Can you retest with that release?

enry86cami, 2022-12-01:
> Code change wasn't pushed out as a release yet. Release now available (V1.7.0) @enry86cami Can you retest with that release?

updated, I let you know....

enry86cami, 2022-12-03:
now when the batteries are 0 the communication stay up

closed 2022-12-04 by binsentsu