binsentsu/home-assistant-solaredge-modbus #126: Meter reported wrong values

ChristophCaina, 2022-12-05:
For Long Time, the Integration of my Inverter worked well and without any issues.

But today, when I wanted to Check the Reports of the Energy usage, I've noticed the following issue in my Energy dashboard:

[image]

It appeared, that the import Meter, as Well as the Export Meter reported a wrong value for a short Period of time.

In fact, this should not happen - and the state_class total increasing should be fine - but maybe - such a wrong Report could be avoided by a Check?

ChristophCaina, 2022-12-06:
today, I noticed the same behave again - around 18:15 ...
I could not see anything within the HA Logs.

It appears, that all import and export sensors are showing a "drop" at the same time:

[image]
[image]

for me, it seems as it happens after the recent update... ?!

dick902, 2022-12-06:
I got the same issues as you after V1.7.0 update!
I had rollback to V1.6.0 and the problem gone.
Seems this is related to the V1.7.0 refactor change.

SgtBatten, 2022-12-07:
What are you rolling back? Can't figure out what 1.7.0 you refer to

binsentsu, 2022-12-08:
Rolled back the validation change for now like before. We'll need to investigate this issue further to see if we can filter it.

ChristophCaina, 2022-12-08:
updated to 1.7.1 and keeping an eye on it :)
Glad, that it became visible while running on my evaluation database :D

TheLordVader, 2022-12-09:
Updated to 1.7.1 and no errors for 24h, just like it was before 1.7.0.

ryanm101, 2022-12-10:
But this makes no sense.. 

Why would a zero reading not be valid?
If my inverter is not producing 0 is the valid state. 

Also changing the battery line `battery_SoC = validate(decoder.decode_32bit_float(), ">=", 0.0)` will now cause the integration to not load at all when the battery is empty.

binsentsu, 2022-12-10:
Think problem is that 0 is also reported as 'default' value in case of connection problems etc. I think for meters and inverter the exported, imported and acenergy are total increasing numbers. Missing in these sensors the first 0.1 is less in my opinion not really a problem. As for the battery, that's maybe another thing as those zeroes are there a commonality.

ChristophCaina, 2022-12-10:
> Think problem is that 0 is also reported as 'default' value in case of connection problems etc. 
That's probably the case - and wouldn't it make sense, that - in case of a disconnect or other 'error' the modbus data won't be read at all?
That would prevent any zero-values due to a connection issue - and, maybe even other false readings...

But yes, the fact that the total value comes back after the connection has been established indicates, that the meter is reporting total increasing values for the export / import / production kwh.
It is only the AC Power / DC Power from the Inverter and Meter that are reporting current production values - and they can report zero if there is no production.

The other sensors are production counters that should always stay with their last value and just sum up.

SgtBatten, 2022-12-10:
> The other sensors are production counters that should always stay with their last value and just sum up.

These summation sensors should not be validated with >= 0 only >0 for that reason. Either it actually is 0 because it's brand new, or every other valid reading that it ever gets will be >0

closed 2023-01-25 by ChristophCaina