binsentsu/home-assistant-solaredge-modbus #129: Document that some inverter by default use id 2

john8329, 2022-12-12:
I'm posting because this took me days of troubleshooting, there's a few non-intuitive things that if addressed have the chance to help a lot of users:

• Logging is nonexistent if readings are unknown, please improve debug logs of abnormal conditions, it's the first place where we look
• My inverter for some reason is responding to id 2 instead of the default 1, this is not documented, and the only reference I found to this id is by configuring a SunSpec logger in the unused RS485-2 interface. Still it works even if I then disable that interface, this is confusing, but it appears that configuration has an impact on the RS485 over TCP connection
• Thanks so much for the info about connecting without SetApp, that's incredibly useful, but please add as a troubleshooting procedure that users should test with `telnet` if the connection works, and the most important information that the inverter accepts only one connection at a time, meaning that subsequent look like the port is closed, but it's not

Cheers

ChristophCaina, 2022-12-18:
Hi,
there's a Discussion board and a Wiki page where such information can be added from everyone :)
I think, there are so many different options that could cause issues - and most of them are depending on the individual setup, that it's pretty hard to write a (nearly) complete trouble-shooting guide.
Especially, if someone hasn't faced any issues so far it is hard to contribute.

If you have some helpful advice, others could profit from, please share them in the corresponding sections, either the wiki or the doiscussions (which could then pinned as important)

john8329, 2022-12-19:
I'd love so, but I posted an issue for two reasons:
• Logging requires editing the code and making it tell when it reaches error conditions
• Most critical information should be put in the README, that's where users land and get the most important instructions, and also there's no link to the wiki. For just a few extra steps I find it more appropriate to put them there instead of a full documentation.

Thanks